ARMmbed/mbed-os #3554: "USBMouse.h" , there is no code for stm32F103

DrYong, 2017-01-10:
Error: #error directive: "Unknown target type" in "USBDevice/USBDevice/USBEndpoints.h", Line: 55, Col: 3

## Question
Could someone finish it?

0xc0170, 2017-01-10:
cc @bcostm @adustm @LMESTM @jeromecoutant

DrYong, 2017-01-31:
//if you compile the following code for NUCLEO-F103, online IDE, there will be #error "Unknown target type"
//F103 is used widely, could 

#include "mbed.h"
#include"USBMouse.h"
USBMouse mouse;
int main()
{
    while (1)
    {
        mouse.move(20,20);
        mouse.click(MOUSE_LEFT);
        wait(1);
    }
}

//The RTC datatime and interrupt function don't work either. For example, the following code to test interrupt will fail for NUCLEO-F103.

//*********not work for stm32f103

#include "mbed.h"
#include "NVIC_set_all_priorities.h"

DigitalOut led1(PC_13);
DigitalOut led2(PC_14);

InterruptIn btn(PA_3);
Serial pc(PA_9,PA_10);

void btnflip()
{
    led1=!led1;
    wait(10);
    led1=!led1;
}

void uartflip()
{
    led2=!led2;
    char c = pc.getc();
    wait(2);
    led2=!led2;
    pc.printf("%c  \n",c);
}

int main()
{
    pc.baud(9600);
    btn.mode(PullUp);
    btn.fall(&uartflip);

    __disable_irq();
    wait(3);
    __enable_irq();

    btn.fall(&btnflip);
    pc.attach(&uartflip);
    NVIC_SetPriority(UART0_IRQn, 5);
    //NVIC_SetPriority(EINT3_IRQn, 255);
    //NVIC_SetPriority
    while (1) {
        printf("xxxxxxxxxxx  \n");
        NVIC_set_all_irq_priorities(255);
         wait(3);
    }
}

jeromecoutant, 2017-07-13:
USB device is supported for NUCLEO_F103RB since #3684

jeromecoutant, 2017-07-13:
ST_TO_BE_CLOSED

closed 2018-01-24 by 0xc0170