ARMmbed/mbed-os #3557: Makefile export should be able to build for multiple profiles

janjongboom, 2017-01-10:
### Description

- Type: Enhancement

## Enhancement

If we export an mbed OS 5 program to (f.e.) Eclipse it does not behave as a 'normal' Eclipse program because our makefile can only support a single profile. Normally, I'd have the ability to do a 'Release' or a 'Debug' build from the IDE, but this is not possible with mbed OS 5 right now. We should change our makefile's so we can do builds for all supported profiles (e.g. `make profile=debug`) so we can change our exporters to support multiple build profiles.

theotherjimmy, 2017-01-10:
Ah, this is a good idea. It might take a lot of work though.

janjongboom, 2017-01-10:
@theotherjimmy Looking at the profiles, shouldn't be too crazy right? Just generate the same Makefile three times (one per profile), but wrapped in ifeq() calls... Not very efficient, but it's generated code anyway.

theotherjimmy, 2017-01-10:
The problem is that the exporter architecture only provides one toolchain with one profile.

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

janjongboom, 2017-10-30:
@ChiefBureaucraticOfficer This is still an issue.

theotherjimmy, 2017-10-30:
@sg- Are we planning to do this?

sg-, 2017-10-30:
You can tell me. If not lets say so and document the workflow and use case. Not just close without resolution.

theotherjimmy, 2017-10-30:
I'll ensure that the docs contain an explanation of `mbed export --profile` and close this when they do.

theotherjimmy, 2018-01-30:
@janjongboom see https://github.com/ARMmbed/mbed-os/pull/5909#issuecomment-361623349