ARMmbed/mbed-os #3570: error return value of osMutexWait/osMutexRelease

kyliuxing, 2017-01-12:
@bulislaw @0xc0170 
We analyze the issue https://github.com/ARMmbed/mbed-os/issues/3435 more and found this is maybe a bug in mbed. SVC call would not be executed immediately, so if we use the return value at once, we may get wrong number form register R0, In this case, it is the mutex id. PLease help check it. thank you!

rt_cmsis.c
osStatus osMutexWait (osMutexId mutex_id, uint32_t millisec) {
if (__get_IPSR() != 0U) {
return osErrorISR; // Not allowed in ISR
}
return __svcMutexWait(mutex_id, millisec);
}
lwip_sys_arch.c
sys_prot_t sys_arch_protect(void) {
if (osMutexWait(lwip_sys_mutex, osWaitForever) != osOK)
error("sys_arch_protect error\n");
return (sys_prot_t) 1;
}
our work around:
sys_prot_t sys_arch_protect(void) {
	osStatus res;
	res = osMutexWait(lwip_sys_mutex, osWaitForever);
	__asm{
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
        nop
    }
    if (res != osOK)
        error("sys_arch_protect error\n");
    return (sys_prot_t) 1;
}

0xc0170, 2017-01-12:
You could have reopened the previous issue as it's the same. Anyway, can you create a simple test case that we can reproduce ?

kyliuxing, 2017-01-12:
will discussed in https://github.com/ARM-software/CMSIS/issues/103, now closed it

closed 2017-01-12 by kyliuxing