ARMmbed/mbed-os #3572: mbed storage: FAT32 POSIX file API defects/bugs/missing functionality

simonqhughes, 2017-01-12:
The following is the current list of defects and missing functionality found from:
- re-instating the FAT32 functionality
- implementing test cases
- feedback provided by the PAL team (Daniel Benor, Clive Bluston

helmut64, 2017-01-13:
I filed a separate errno enhancement see #3589

helmut64, 2017-01-13:
At present there are major memory problems in the FAT32 FileSystem subsystem. There is too much data is located on the Stack e.g.:
- FATFS structure one sector 512 bytes being on the stack
- FIL structure per file one sector 512 bytes plus temporary another 512 bytes per FIL

This means more than 1.5kB which is way to much and results into stack overflows with hidden problems appearing later somewhere else.

clivebluston, 2017-01-15:
In the following scenario:
errno is getting set to 9 by fwrite. I do not think that this should happen.

    FILE *f = fopen("/sd/test.txt","w+");
    if(!f)
    {
    	printf("file system error\n");
    }
    else
    {
    	errno=0;
    	clearerr(f);
    	int res = fwrite("hello",6,1,f);
    	printf("fwrite res=%d, errno=%d ferror=%d\n",res,errno,ferror(f));
    	fclose(f);
    }
Output:
fwrite res=1, errno=9 ferror=0

helmut64, 2017-01-15:
FYI: The errno should only be evaluated when the fwrite returns error. e.g. fwrite() == 0.

helmut64, 2017-01-15:
Some additional feedback for the FATFileSystem.
- The static data for LfnBuf[_MAX_LFN + 1] and  sfn[12] should be dynamically allocated to avoid the memory use when the FATFileSystem is only used used temporary.
- The _MIN_SS/_MAX_SS max sector size configuration should be parameters in the constructor. This allows to dynamically allocate the memory and to adjust the sector size for the requested needs. E.g. I use _MIN_SS=4k_MAX_SS=4k to match the sector size of the underlying NOR flash memory.
- The lower level (open/read/write) system calls are not available in the mbed file system switch. The use of stdio fread/write is not always appropriate due to its own stdio buffering which eats memory 

I did some enhancements for my own project, the status of the FATFileSystem is not fully clear to me because it is located in the features/unsupported directory, therefore I have not considered to deliver patches.

Regards Helmut

clivebluston, 2017-01-16:
Helmut64. I am not sure if errno should get set by fwrite when there is an error. My man page on fwrite does not mention setting errno, though it does mention functions ferror(),clearerror and feof(). I am not sure if it is a mistake in the man page or not. In any case it is setting errno even when no error occurs which can lead to issues in subsequent code.

clivebluston, 2017-01-16:
Helmut64. Simon Hughes should be able to make a statement regarding the status.

helmut64, 2017-01-16:
Dear Clive, the errno is not set in the fwrite, it is set the the low level write function which should receive the error code from the filesystem. The call chain is will somwhow look like this:
App-> fwrite->write->semihost-syscall->filesystem->BlockDevice

The filesystem/BlockDevice will set the errno and the higher level can use it.

simonqhughes, 2017-01-20:
Just wanted to give you a further update on today’s work:

- STORAGE_ISSUE_0001: errno not working. 
    - Basic errno reporting should now be working. EEXIST is returned from mkdir() on pre-existing file. All file handing functions have errno set if an error occurs.
- STORAGE_ISSUE_0002: ftell (possibly) functionality not present. 
    - This is working and tested.
- STORAGE_ISSUE_0003: dirent doesn't expose file/directory type information
    - Whether an object is a file or type can be determined with stat(). Test case not present yet.
- STORAGE_ISSUE_0004: freopen() doesn't close fopen() file description.
    - Not progressed.
- STORAGE_ISSUE_0005: tmpnam() doesnt currently generate a temporary filename
    - Not progressed.
- STORAGE_ISSUE_0006: STDIN_FILENO, STDOUT_FILENO, STDERR_FILENO not available … 
    - Not progressed.
- STORAGE_ISSUE_0007: PAL team required core functionality not tested and quality unknown
    - Ongoing. More test cases being implemented.
    - The core set of functionality as described under this point is usable.
- STORAGE_ISSUE_0008: File system wrappers use new for dynamic memory allocation on the heap (poor …
    - Not progressed.

helmut64, 2017-01-22:
@simonqhughes once you have checked in your changes I will apply my enhancements. At present I use a local copy of the FATFileSystem. A proper working FAT would enhance mbed for all of us.

simonqhughes, 2017-02-08:
Just wanted to give you an update on current storage status.

- STORAGE_ISSUE_0001: errno not working. 
    - The full errno reporting has now been implemented in this PR: https://github.com/ARMmbed/mbed-os/pull/3704. 
    - Please can you confirm this issue can be marked as CLOSED?
- STORAGE_ISSUE_0002: ftell (possibly) functionality not present. 
    - This is working and tested. There is a test case in fopen.cpp:: fsfat_fopen_test_09().
    - Please can you confirm this issue can be marked as CLOSED?
- STORAGE_ISSUE_0003: dirent doesn't expose file/directory type information
    - The fix has been merged into feature-storage branch with this PR: https://github.com/ARMmbed/mbed-os/pull/3720. There is a new test case operating on ram driver in fat_file_system/main.cpp. You may like to test on sd card.
    - Please can you confirm this issue can be marked as CLOSED?
- STORAGE_ISSUE_0004: freopen() doesn't close fopen() file description.
    - Not progressed.
- STORAGE_ISSUE_0005: tmpnam() doesnt currently generate a temporary filename
    - Not progressed.
- STORAGE_ISSUE_0006: STDIN_FILENO, STDOUT_FILENO, STDERR_FILENO not available … 
    - This has been fixed here: https://github.com/ARMmbed/mbed-os/pull/3704. 
    - This can be marked CLOSED now.
- STORAGE_ISSUE_0007: PAL team required core functionality not tested and quality unknown
    - Ongoing. From discussions with Clive/Jenia on Monday this week, I understand the core functionality is working OK. 
    - More test cases being implemented.
- STORAGE_ISSUE_0008: File system wrappers use new for dynamic memory allocation on the heap
    - Not progressed. 
    - Not intending to progress at this time.

helmut64, 2017-02-08:
Thanks for the Update, I will look into it.

closed 2017-10-27 by ChiefBureaucraticOfficer