ARMmbed/mbed-os #3576: STM32L4 IWDG WatchDog cannot be used in mbed due to a problem in rtc_init/rtc_api.c

helmut64, 2017-01-12:
@bcostm 
Note: This is just a template, so feel free to use/remove the unnecessary things
- Type: Enhancement

Enabling the IWDG WatchDog aborts mbed in rtc_init with  the error message “Cannot initialize RTC with LSE”.

The problem is, that the init_rtc aborts when setting the mbed time via set_time(t). The IWDG enables the LSE because it works even in deepsleep mode when the clock is off, however the rtc_init() turns off the LSE therefore the HAL_RCC_OscConfig() fails.

It would be good if this limitation can be fixed, I don’t understand all details about the clock therefore someone with more STM know how better looks into it.

Regards Helmut

PS: I can provide a sample code to enable the IWDG, however I belive it is not needed.

bridadan, 2017-01-13:
Thanks for reporting @helmut64! Any sample code that you're using to reproduce this problem is useful as it helps us debug the problem faster, so feel free to post it!

cc @jeromecoutant

helmut64, 2017-01-13:
Believe or not the problem disappears in  build #133 in #132 it still exists, I have not switched to 133 for some power usage reason which is fixed already (upcoming 134 release).

My sample code here: (not needed anymore)
https://developer.mbed.org/users/Helmut64/code/STM32L4_WatchDog/

I close this issued because it is obsolete in revision 133.My sample code here:
Regards Helmut

closed 2017-01-13 by helmut64

bridadan, 2017-01-13:
Ah good to hear! Thanks for following up :smile: