ARMmbed/mbed-os #3581: ARM and uARM build broken by mbed-os-5.3.2 release

toyowata, 2017-01-13:
### Description
- Type: Bug
- Related issue: 
- Priority: Blocker

---------------------------------------------------------------
## Bug

mbed classic build tool seems to be broken between mbed-os-5.3.1 and 5.3.2 release.
I got build fail [OS error] by ```-m LPC1768 -t ARM``` build.

**Target**
LPC1768 and other classic targets

**Toolchain:**
ARM and uARM

**Toolchain version:**
Product: MDK Professional 5.22
Component: ARM Compiler 5.06 update 4 (build 422)
Tool: armcc [4d3604]

**mbed-cli version:** 
Not use

**Expected behavior**
Build success

**Actual behavior**
Build fail

**Steps to reproduce**

[mbed-os-5.3.2 release]

```
X:\mbed-os>python tools\build.py -m LPC1768 -t ARM
Building library CMSIS (LPC1768, ARM)
Scan: cmsis
Building library MBED (LPC1768, ARM)
Scan: drivers
Scan: platform
Scan: hal
Scan: targets
Scan: TARGET_LPC1768
[OS ERROR] Command: bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M3 --md --no_depend_system_headers --c99 -D__ASSERT_MSG -DDEVICE_ERROR_PATTERN=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTA
RGET_NXP -DTARGET_LPC176X -D__MBED_CMSIS_RTOS_CM -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=
1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DMBED_BUILD_TIMESTAMP=1484295786.34 -DTARGET_LPC1768 -DTARGET_RELEASE -DDEVICE_PORTINOUT=1 -DDEVICE_SERIAL_FC=1 -DTARGET_MBED_LPC1768 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -D
DEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_SEMIHOST=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DDEVICE_LOCALFILESYSTEM=1 --via X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\.includes_ffabbaafb430829311a6fbc10dfa5948.txt --preinclude=X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_A
RM_STD\mbed_config.h --depend X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\analogin_api.d -o X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\analogin_api.o X:\mbed-os\targets\TARGET_NXP\TARGET_LPC176X\analogin_api.c
[Error 2] 指定されたファイルが見つかりません。

Completed in: (0.44)s

Build failures:
  * ARM::LPC1768
[OS ERROR] Command: bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M3 --md --no_depend_system_headers --c99 -D__ASSERT_MSG -DDEVICE_ERROR_PATTERN=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTA
RGET_NXP -DTARGET_LPC176X -D__MBED_CMSIS_RTOS_CM -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=
1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DMBED_BUILD_TIMESTAMP=1484295786.34 -DTARGET_LPC1768 -DTARGET_RELEASE -DDEVICE_PORTINOUT=1 -DDEVICE_SERIAL_FC=1 -DTARGET_MBED_LPC1768 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -D
DEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_SEMIHOST=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DDEVICE_LOCALFILESYSTEM=1 --via X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\.includes_ffabbaafb430829311a6fbc10dfa5948.txt --preinclude=X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_A
RM_STD\mbed_config.h --depend X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\analogout_api.d -o X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\analogout_api.o X:\mbed-os\targets\TARGET_NXP\TARGET_LPC176X\analogout_api.c
[OS ERROR] Command: bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M3 --md --no_depend_system_headers --c99 -D__ASSERT_MSG -DDEVICE_ERROR_PATTERN=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTA
RGET_NXP -DTARGET_LPC176X -D__MBED_CMSIS_RTOS_CM -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=
1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DMBED_BUILD_TIMESTAMP=1484295786.34 -DTARGET_LPC1768 -DTARGET_RELEASE -DDEVICE_PORTINOUT=1 -DDEVICE_SERIAL_FC=1 -DTARGET_MBED_LPC1768 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -D
DEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_SEMIHOST=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DDEVICE_LOCALFILESYSTEM=1 --via X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\.includes_ffabbaafb430829311a6fbc10dfa5948.txt --preinclude=X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_A
RM_STD\mbed_config.h --depend X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\can_api.d -o X:\mbed-os\BUILD\mbed\.temp\TARGET_LPC1768\TOOLCHAIN_ARM_STD\TARGET_NXP\TARGET_LPC176X\can_api.o X:\mbed-os\targets\TARGET_NXP\TARGET_LPC176X\can_api.c

X:\mbed-os>
```

[mbed-os-5.3.1 release]

```
Y:\mbed-os>python tools\build.py -m LPC1768 -t ARM
Building library CMSIS (LPC1768, ARM)
Scan: cmsis
Copy: core_cm0.h
Copy: core_cm0plus.h
Copy: core_cm3.h
Copy: core_cm4.h
Copy: core_cm7.h
Copy: core_sc000.h
Copy: core_sc300.h
Building library MBED (LPC1768, ARM)
Scan: drivers
Scan: platform
Scan: hal
Scan: targets
Copy: mbed_lib.json
Scan: TARGET_LPC1768

(snip)

Compile [ 93.2%]: mbed_stats.c
Compile [ 95.5%]: mbed_wait_api_rtos.cpp
Compile [ 97.7%]: mbed_wait_api_no_rtos.c
Compile [100.0%]: retarget.cpp
Library: mbed.ar
Copy: mbed_board.o
Copy: retarget.o

Completed in: (11.93)s

Build successes:
  * ARM::LPC1768

Y:\mbed-os>
```
Other classic build target also broken such as WIZWIKI_W7500 (ARM), RZ_A1H (ARM), LPC1114 (uARM) etc.

0xc0170, 2017-01-13:
Hi,

``python tools\build.py -m LPC1768 -t ARM``

master - builds OK
mbed-os-5.3 branch - builds OK
mbed-os-5.3.2 tag - builds OK

5.3.2 - 
```
`*   2885c1b - (HEAD, tag: mbed_lib_rev133, tag: mbed-os-5.3.2, origin/release-candidate, origin/mbed-os-5.3) Merge pull request #3510 from A|\
| * 0c4ccdc - Update MBED_LIBRARY_VERSION to v133. Update mbed-os 5 patch version. (2 weeks ago) <Anna Bridge>
| * 14a2e3a - Rename KSDK2 to MCUXpresso. This is the new name of this package from NXP (2 weeks ago) <Mahadevan Mahesh>

Build successes:
  * ARM::LPC1768

```

0xc0170, 2017-01-13:
How to reproduce your failures? What failure you get, I can't read the error status.

btw, I removed BUILD folder everytime I switched branches/tags.

adbridge, 2017-01-13:
I just built blinky using LPC1768 as the target and the latest version of mbed (v133) and it built fine...

toyowata, 2017-01-13:
Hi @0xc0170 and @adbridge 
Thank you very much for your response.

>How to reproduce your failures? What failure you get, I can't read the error status.

Step to reproduce
```
> git clone -b mbed_lib_rev133 https://github.com/ARMmbed/mbed-os 
> cd mbed-os
> python tools\build.py -m LPC1768 -t ARM
```
The [OS ERROR] message (in Japanese):
```
[Error 2] 指定されたファイルが見つかりません。
```
In English:
```
[Error 2] Cannot find the file specified.
```

It seems the command-line wrongly called ```bin\armcc``` instead of ```armcc``` with full path.

I just found difference between rev133 and rev132 in tools\settings.py and the ARM_PATH is not same.

[rev132]
```
# ARM Compiler 5
ARM_PATH = "C:/Keil_v5/ARM/ARMCC"
```

[rev133]
```
# ARM Compiler 5
ARM_PATH = ""
```

I did have ARM_PATH in my system environment variable which is same as settings.py name, but it was not used for build.  Also, I did not have MBED_ARM_PATH environment variable.

```
C:\Users\toywat01\Documents\mbed\test\mbed-os>set ARM_PATH
ARM_PATH=C:\Keil_v5\ARM\ARMCC
C:\Users\toywat01\Documents\mbed\test\mbed-os>set MBED_ARM_PATH
Environment variable MBED_ARM_PATH not defined
```

When I add MBED_ARM_PATH and set it to the correct path location, it works fine in rev133 (build.py builds mbed library without error).

0xc0170, 2017-01-13:
cc @bridadan - the commiter of that change

From the commit:
>   This change proposes to remove this default behavior and instead
    requires the user to explicitly set the intended compiler, either by a
    settings file, mbed CLI, environment variables, or by placing the
    compiler in your PATH.

bridadan, 2017-01-13:
You can find the full reasoning (and the issue that motivated me to make the change in the first place) in the PR here: https://github.com/ARMmbed/mbed-os/pull/3476

The script should have realized you didn't have a valid toolchain executable and given you a nice message instead of just a `[OS ERROR]`. So I went ahead and reproduced the issue locally. And the issue is here: https://github.com/ARMmbed/mbed-os/blob/master/tools/build.py#L217. That line shouldn't be in the `if` block, it needs to be moved.

Here's the PR that should fix it: https://github.com/ARMmbed/mbed-os/pull/3587

toyowata, 2017-01-14:
@bridadan Thanks for the PR https://github.com/ARMmbed/mbed-os/pull/3587 
I tested this locally and confirmed the fix.  I could build mbed library without OS ERROR.  Thanks!

PR #3587 (2017-01-26): Fixing toolchain executable not found error for build.py
## Description
Commit 19d56fd40fc3130b970e03fd530ab340d4a65465 removed the default file
paths for the toolchains. This was done under the assumption that the
top-level compile scripts were properly checking that the toolchain
executable was availble. The build.py script was doing this in the wrong
place. This commit rearranges the script a bit so the check is performed
properly.

Should fix #3581.

## Status
**READY**

## Migrations
If this PR changes any APIs or behaviors, give a short description of what *API users* should do when this [... cut]
closed 2017-01-16 by 0xc0170