ARMmbed/mbed-os #3582: Broken InterruptIn() for SAMD21 target when using mode() to change Pullmode

finneyj, 2017-01-13:
When using a SAMD21 target, GPIO interrupts registered with InterruptIn() stop firing when the PullMode is subsequently configured using the mode()  function.

I tracked this down to the following line, which is called from mode():
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/drivers/system/pinmux/pinmux.c#L75

Note that if the pin is in GPIO mode, the PMUXEN bit is overwritten. This bit should be set to '1' to map the pin onto the EIC module that performs edge detection etc. Essentially a bad assumption that if the GPIO pin is in GPIO mode, then the PMUXEN mode doesn't matter (which is not true).

For reference, I have a workaround in a local branch. Please see commit below:
https://github.com/lancaster-university/mbed-classic/commit/645e7e086a4b126728a11cc1824ebbdab1b4fe70

bridadan, 2017-01-13:
@finneyj Thanks for tracking down this bug! Would you mind submitting your change as a Pull Request? We'd love to have the fix! :smile: 

Though before you submit it, would you mind just deleting the commented-out lines of code?

finneyj, 2017-01-13:
Hey, no worries!

Sure thing. I don't think the fix is quite right though - close, but I'm not sure it covers all the bases....

theotherjimmy, 2017-01-13:
That's fine @finneyj. We can work out the problems faster if we have a way of tracking it. Further, submitting a PR allows us to run our CI jobs against it. This can quickly show the corner cases that are not working, supposing there are any.

0xc0170, 2017-01-16:
cc @Parthasarathy @kpurusho

closed 2017-10-27 by ChiefBureaucraticOfficer