ARMmbed/mbed-os #3589: Usefull errno.h and errno() subsystem missing in mbed

helmut64, 2017-01-13:
- Type:   Enhancement 

Multiple subsystems would benefit from a central errno support in mbed. I understand that there would be an immediate benefit for lwip and FATFileSystem and for others to avoid duplicate code and to enhance error handling.

A central errno definition and a errno variable per thread would have only a few bytes more data and little code size. To avoid large strerror string tables I would recommend to skip the strerror string table or to add a dummy just returning the errno number.

I filed this topic for discussion.

Regards Helmut

PS: I can handle the developments of errno support (if there is consensus  about it)

bridadan, 2017-01-13:
@helmut64 I definitely agree that the error reporting within mbed OS could use some consolidation and clarification. Same thing for our different logging/debug output.

@sg- or @0xc0170 do you have any thoughts on this?

@geky you may have some thoughts on this from your work with lwip and the FATFileSystem.

geky, 2017-01-13:
I was just thinking about this. I agree, we need a better standardization of error handling. Current error handling falls a bit short now that we are seeing complex libraries with different layers of failures.

However, I'm hesitant to adopt errno for use with the C++ apis short of posix compatibility. errno doesn't keep track of the context of the originating error and can lead to tricky user-facing problems. Even simple snippets of code can deceptively misuse errno:
``` cpp
errno = 0;
int size = socket->send(buffer, buffer_size);
printf("sent: %d\r\n", size);
if (size != buffer_size) {
    printf("error: %d\r\n", errno);
}
```

The current standard is to simply return an error code if an error occurs. But this raises the question of what to do when an API needs to return a value:
``` cpp
int error = mx25r->read_sfdp(&sfdp);              // ?
uint64_t sfdp = mx25r->read_sfdp(&error);         // ?
Tuple<uint64_t, int> result = mx25r->read_sfdp(); // ?
```

The more serious concern is how easy it is for users to ignore errors. Look at our higher-level examples and it's easy to find code that continues happily after deadly errors.

One solution is to adopt sticky errors on a per-class basis. This removes the contextless problem of errno, and provides a simple mechanism for coalescing errors in this exceptionless world. In the future we could even have the classes log out any errors that aren't handled.
``` cpp
int size = socket->send(buffer, buffer_size);
printf("sent: %d\r\n", size);
if (size != buffer_size) {
    printf("error: %d\r\n", socket->get_error());
}
```

---

There is another possible solution, but this might be too fantastical for adoption in mbed. We could integrate a first-class result object that wraps any error-dependent return values, and overload the implicit cast to log out any errors (or assert) if the error was never handled:
``` cpp
// Handle error correctly
Result<int> result = socket->send(buffer, buffer_size);
if (!result) {
    printf("error: %d\r\n", result.get_error());
}
printf("sent: %d\r\n", result.get_result());

// Don't handle error, mbed will log error or assert if an error occurs
int size = socket->send(buffer, buffer_size);
printf("sent: %d\r\n", size);
```

Do let me know if anyone is interested in this route, it may be worth persuing for the extra garuntee it has on error handling.

helmut64, 2017-01-14:
I agree that sometimes it is not clear where the errno comes from, e.g.:
fopen() == NULL
- could be a out of memory in fopen code
- invalid path in the OS file system switch
- no permission in the filesystem.
- IO error in the flash driver.

However it is also good to keep in mind that errno works fine on Linux, Solaris, Mac OS, AIX, Windows (via GetLastError lime _errno()) always the same way. This means users would benefit from identical behaviour in mbed. Addeding an errno stack would be better but is also more difficult to implement and use.

A complete different approach would be to implement error codes it via C++ exception objects which can automatically unwind entire calling chains (via cleanup in destructors) if not catched. The disadvantage of this C++ exception is that the HAL code is mostly regular C where errno would be a better fit.

For IO functions I write, I do it mostly like this:
`
if (socket->send() == -1)
	handle error; // no printf before we because this could overwrite the error.
`

pan-, 2017-01-16:
> There is another possible solution, but this might be too fantastical for adoption in mbed. We could  integrate a first-class result object that wraps any error-dependent return values, and overload the implicit cast to log out any errors (or assert) if the error was never handled:

> ```c++
> // Handle error correctly
> Result<int> result = socket->send(buffer, buffer_size);
> if (result) {
>     printf("sent: %d\r\n", result.get_result());
> } else {
>     printf("error: %d\r\n", result.get_error());
> }
> 
> // Don't handle error, mbed will log error or assert if an error occurs
> int size = socket->send(buffer, buffer_size);
> printf("sent: %d\r\n", size);
> ```

> Do let me know if anyone is interested in this route, it may be worth persuing for the extra garuntee it has on error handling.

@geky I've used that kind of error handling in the past it is very handy if you know what you are a doing even on system using exception. There is a good presentation of the concept [here](https://channel9.msdn.com/Shows/Going+Deep/C-and-Beyond-2012-Andrei-Alexandrescu-Systematic-Error-Handling-in-C) and a C++ proposal [here](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2014/n4109.pdf). Returning an arbitrary result type or an arbitrary error type is a big semantic improvement over integer codes and when you want to avoid `output` parameters in function signature. 

Such construct also tends to make the code of the function returning the error more readable if the right functions are in place: 

```C++
Expected<size_t, int> send(const void* buffer, const size_t buffer_size) { 
   size_t result = 0;

   if (!is_connected) { 
      return make_error(ENOTCONN); 
   }

   // continue to return errors with make_error

   // return the result like it is a regular value 
   return result;
}
```

But even if syntactic sugar is appealing it comes with a cost in term of memory and CPU consumption because mbed is a C++98 codebase and move semantic is not available. I also wants to point out that users should be knowledgeable of C++ (not just C or C with class) to avoid the gotchas induced by this approach especially regards to RAII/non RAII types. 

More importantly developers feels like they are in familiar ground when they see functions like `open`, `read`, `write`, `bind`, etc. They expect those functions to behave and report error a certain way. It would increase the barrier entry and raise incomprehension over mbed to change the error handling mechanism of these primitives especially when they are taken for granted. Implementing correctly `errno` variable would on the other hand reinforce mbed "compliance" with other well known OS. 

This raise another question: Should there be a single error signaling mechanism ? `errno` is an artifact of the past easy to misuse. It still makes a lot of sense for functions of the POSIX interface but certainly not for its error reporting or error handling quality.  Would it makes sense to use another way to report  and handle error for C++ based code ?

helmut64, 2017-01-16:
I just browsed the mbed/targets directory where properly most error code handling will come from, there is not a single C++ source file from the target vendors. So an classic errno per thread would do the job. I would also recommend to reserve error code range per vendor so that every vendor can add their error numbers when needed. E.g.:
- errnos 1-10000 for mbed and generic errnos
- 10k error ranges for each vendor and/or library.

One the other hand I don't mind if a nice C++ implementation gets done.

pan-, 2017-01-17:
@helmut64 `errno` values are well defined (see [here](http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/errno.h.html)). I believe it would be a mistakes to extend it on something else because it is not expected by the user. Vendor specific error code would also makes mbed programs less portable across different boards.

@sg- @0xc0170 Could you share your thought around error handling strategy in mbed ?

helmut64, 2017-01-17:
@pan- The well defined errnos may not be sufficient, check the Windows errnos, Linux errnos, OS X errnos and Solaris errnos, than you will find our that this is a moving target.

I pretty much like to be able that vendors/libraries can define in addition their own errnos to allow to report hardware failures more precisely in case it cannot be mapped to a standard errno. E.g. certificate expired, etc.

Thanks for your input as links.

pan-, 2017-01-17:
@helmut64 On Windows, only a subset and not a superset of posix errno values is used. [System error code](https://msdn.microsoft.com/en-us/library/windows/desktop/ms681382(v=vs.85).aspx) can be accessed by the function [`GetLastError`](https://msdn.microsoft.com/en-us/library/windows/desktop/ms679360(v=vs.85).aspx). Is errno used on POSIX systems to reports vendor error code ?

helmut64, 2017-01-17:
@pan- Yes and No.
Lets say if Mac OS, IBM AIX, Oracle Solaris bring out a new version, they extent their errno.h list regulary.

In our company software we use hundreds of additional error codes which we added in our libraries and our own strerrnor() function returns the system OS error strings as well our library error strings depending on the errno range. Certainly the strerrnor() function I would skip for mbed due to flash sizes. However I am confident that not every HAL error code maps into a valid/good errno, therefore allowing each vendors/libraries to add their own numbers in a way that they don't overlap would be reasonable.

Anyways just collecting ideas.

closed 2017-10-27 by ChiefBureaucraticOfficer