ARMmbed/mbed-os #3593: Exporting to e2 studio will result in build error

TomoYamanaka, 2017-01-17:
### Description
- Type: Question
- Priority: Major 

---------------------------------------------------------------
## Question
When exporting the existing mbed classic program such as mbed_blinky to Renesas' IDE **e2studio**, the following error is shown in building the exported project:<br>

`mbed_blinky_e2studio_rz_a1h\mbed_blinky\mbed/drivers/Serial.h:70:70: error: 'MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE' was not declared in this scope`

It seems that `mbed_config.h` should be manually included at present so that we can build the project.
We believe this should **NOT** be the expected behavior. **Could you confirm the above?**

0xc0170, 2017-01-17:
cc @theotherjimmy  @sarahmarshy

theotherjimmy, 2017-01-17:
@TomoYamanaka You are correct! It really should not do that. e2studio was not on my road map, but it is now. Is e2studio an eclipse based IDE? if so, we can easily use eclipse_gcc_arm as a replacement for now.

TomoYamanaka, 2017-01-18:
@theotherjimmy 

Thank you for your comments.
For now, I will try with the above replacement method.
Please contact me if you know the support schedule  etc of this issue.

TomoYamanaka, 2017-03-09:
This issue traces with #3889.

TomoYamanaka, 2017-03-14:
@phyokyaw

This issue is a same cause #3889, is traced with here.
I guess that you are dealing with this issue, and you are modifing the template file for export to e2 studio (ex: rz_a1h_cproject.tmpl).

So we provied the file to referece for correction.
After then, how about the status of this issue ?

TomoYamanaka, 2017-03-23:
This issue modified and merged in PR of #3933. Therefore I close this. The isseue of #3889 also close.

TomoYamanaka, 2017-04-19:
Please let me re-open this issue since I faced similar issue.

I confirmed the e2studio project template was actually merged with mbed OS 5.4.2.
However, it seems that the same error is seen when exporting mbed_blinky to e2studio project on **online compiler**.

Also, no error is shown when using **mbed export** command to export the project.

Could you check if rz_a1h_cproject.tmpl included in mbed OS 5.4.2 is also used on online compiler?

sg-, 2017-04-19:
@theotherjimmy

theotherjimmy, 2017-04-19:
I'll update the tools to mbed OS 5.4.2 in the online compiler backend

theotherjimmy, 2017-04-19:
@TomoYamanaka I have submitted a change to update the tools on the online complier backend.

TomoYamanaka, 2017-05-10:
I confirmed that this issue was resolation.
So I close this issue.

closed 2017-05-10 by TomoYamanaka