ARMmbed/mbed-os #3596: New mbed rev 134 fails with error: mbed.h not found.

andcor02, 2017-01-17:
So with the new mbed 2.0 rev 134 .

**Projects will not compile. 

> Error: Cannot open source input file "mbed.h": No such file or directory in "main.cpp", Line: 9, Col: 19

Will work before rev 134.**

Please Advice,

Regards,

Andrea

0xc0170, 2017-01-17:
@andcor02 What target? Any?

cc @adbridge

andcor02, 2017-01-17:
Any Target it seems

adbridge, 2017-01-17:
@andcor02 how are you compiling? (online IDE or using mbed-cli ? ). Are you using the mbed compiled lib or the mbed-dev sources ? 
I just compiled the blinky example for the LPC1768 in the online IDE with the latest mbed compiled lib and it build fine for me ?

adbridge, 2017-01-17:
@andcor02 I should say that 134 hasn't been officially released yet though so maybe there is an issue with the online IDE not picking things up correctly for you..

andcor02, 2017-01-17:
@adbridge 

Compiling with online IDE, using mbed compiled lib.

I just tried what you did and it didn't compile, did you make sure to switch the revision to 134? For me it doesn't automatically update to 134, even when choosing to.

adbridge, 2017-01-17:
@andcor02 I just deleted my version of mbed from my blinky and then imported from the URL https://developer.mbed.org/users/mbed_official/code/mbed/

andcor02, 2017-01-17:
@adbridge 

Something strange is happening here indeed the above does work. 

Seems to be all fixed now, the workflow that was broken for me now works. Strange!

andcor02, 2017-01-17:
Thanks for your help! :D

closed 2017-01-17 by andcor02