ARMmbed/mbed-os #3598: mbed-os-example-client size with UBLOX_EVK_ODIN_W2 is massive

JanneKiiskila, 2017-01-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**
0.9.10

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)
5.3.3

**DAPLink version:**

**Expected behavior**

mbed-os-example-client binary size should match approximately what it is for the rest of the boards, somewhere from 250 to 500 kB. 

**Actual behavior**

The binary is huge, it 1.1 megabytes.

**Steps to reproduce**

git clone mbed-os-example-client
generate security.h from portal (connect.mbed.com)
./build_all.sh
check binary sizes

JanneKiiskila, 2017-01-17:
jankii01@ubuntu:~/mbed/mbed-os-example-client$ ll *.bin
-rwxrwxr-x 1 jankii01 jankii01  404456 Jan 17 14:32 f439zi-GCC_ARM-6lowpan.bin*
-rwxrwxr-x 1 jankii01 jankii01  278752 Jan 17 14:31 f439zi-GCC_ARM-eth-v4.bin*
-rwxrwxr-x 1 jankii01 jankii01  435392 Jan 17 14:33 f439zi-GCC_ARM-Thread.bin*
-rwxrwxr-x 1 jankii01 jankii01  395132 Jan 17 14:32 k64f-GCC_ARM-6lowpan.bin*
-rwxrwxr-x 1 jankii01 jankii01  271516 Jan 17 14:29 k64f-GCC_ARM-eth-v4.bin*
-rwxrwxr-x 1 jankii01 jankii01  274212 Jan 17 14:30 k64f-GCC_ARM-eth-v6.bin*
-rwxrwxr-x 1 jankii01 jankii01  426228 Jan 17 14:33 k64f-GCC_ARM-Thread.bin*
-rwxrwxr-x 1 jankii01 jankii01  237876 Jan 17 14:31 k64f-GCC_ARM-wifi-v4.bin*
-rwxrwxr-x 1 jankii01 jankii01 1147488 Jan 17 14:31 ublox-odin-GCC_ARM-wifi-v4.bin*

JanneKiiskila, 2017-01-17:
What is really strange is that if you look at the mbed-os-example-client-map.csv,  total flash is 464853 -> so the size should not be even remotely close to 1 megabyte? What's the linker doing in the end?

JanneKiiskila, 2017-01-17:
```
mbed compile -m UBLOX_EVK_ODIN_W2 -t GCC_ARM -c
Link: mbed-os-example-client
Elf2Bin: mbed-os-example-client
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Fill                        |    662 |    15 |    86 |
| Misc                        | 286923 |  2653 | 18052 |
| drivers                     |   1255 |     0 |     0 |
| events                      |     64 |     0 |     0 |
| events/equeue               |   1226 |     0 |    61 |
| features/FEATURE_COMMON_PAL |  16651 |    93 |  8345 |
| features/FEATURE_LWIP       |  34802 |    87 | 46801 |
| features/mbedtls            |  90261 |    55 |  8755 |
| features/netsocket          |   5046 |    85 |    60 |
| hal                         |    672 |     0 |     8 |
| platform                    |   1390 |     4 |   313 |
| rtos                        |    757 |     4 |     4 |
| rtos/rtx                    |   7851 |    20 |  6871 |
| targets/TARGET_STM          |  14143 |     4 |   784 |
| Subtotals                   | 461703 |  3020 | 90140 |
+-----------------------------+--------+-------+-------+
```

JanneKiiskila, 2017-01-17:
K64F with ESP8266 as reference:

```
Elf2Bin: mbed-os-example-client
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Fill                        |    453 |    14 |  2297 |
| Misc                        | 117785 |  2521 |  1427 |
| drivers                     |   1659 |     0 |     0 |
| features/FEATURE_COMMON_PAL |  16651 |    89 |  8345 |
| features/mbedtls            |  73371 |    51 |  8755 |
| features/netsocket          |   4082 |    85 |     0 |
| features/storage            |     42 |     0 |   184 |
| hal                         |    580 |     0 |     8 |
| platform                    |   1424 |    20 |   297 |
| rtos                        |    757 |     4 |     4 |
| rtos/rtx                    |   7221 |    20 |  6871 |
| targets/TARGET_Freescale    |   9787 |    12 |   872 |
| Subtotals                   | 233812 |  2816 | 29060 |
+-----------------------------+--------+-------+-------+
Allocated Heap: 65540 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 31876 bytes
Total RAM memory (data + bss + heap + stack): 97416 bytes
Total Flash memory (text + data + misc): 237668 bytes

```

0xc0170, 2017-01-17:
cc @andreaslarssonublox

JanneKiiskila, 2017-01-18:
Issue also exists with mbed-os-example-wifi, so it's not to do with mbed-os-example-client.

For K64F
`80056 Jan 18 09:55 mbed-os-example-wifi.bin*`

UBLOX;
```
Elf2Bin: mbed-os-example-wifi
+-----------------------+--------+-------+-------+
| Module                |  .text | .data |  .bss |
+-----------------------+--------+-------+-------+
| Fill                  |    387 |     8 |    71 |
| Misc                  | 219824 |  2580 | 17165 |
| drivers               |   1068 |     0 |     0 |
| events                |     64 |     0 |     0 |
| events/equeue         |   1226 |     0 |    61 |
| features/FEATURE_LWIP |  34782 |    87 | 46796 |
| features/mbedtls      |  20313 |     0 |  8748 |
| features/netsocket    |   4800 |    85 |    60 |
| hal                   |    490 |     0 |     8 |
| platform              |   1196 |     4 |   269 |
| rtos                  |    213 |     4 |     4 |
| rtos/rtx              |   7279 |    20 |  6870 |
| targets/TARGET_STM    |  13729 |     4 |   844 |
| Subtotals             | 305371 |  2792 | 80896 |
+-----------------------+--------+-------+-------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 83688 bytes
Total RAM memory (data + bss + heap + stack): 83688 bytes
Total Flash memory (text + data + misc): 308163 bytes

990268 Jan 18 09:56 mbed-os-example-wifi.bin*
```

So, should be around 308 kb but it is 990 kb.

JanneKiiskila, 2017-01-18:
Kudos to @teetak01 for actually spotting that this is an issue. I just did some more digging on this and files this bug.

andreaslarssonublox, 2017-01-18:
The u-blox ODIN-W2 Wi-Fi driver takes approximately 1M of flash since the driver contains patch files for Wi-Fi(682k) and Bluetooth in addition to the stacks.
The patch file can be found in the generated wifi.map file:
.firmware 0xa6966(682342)  cb_fw_owl35x.o

This is defined as:
static const cb_uint8 __attribute__((section (".firmware"))) fw[] = { .... }

When building with ARM and IAR toolchains the numbers look OK:
| wifi                              | UBLOX_EVK_ODIN_W2 | ARM       |      84004 |     0 |    0 |     84004 |     1038251 |
| wifi                              | UBLOX_EVK_ODIN_W2 | IAR       |      83645 | 24576 | 49152 |    157373 |      984678 |
while GCC_ARM is:
| wifi                              | UBLOX_EVK_ODIN_W2 | GCC_ARM   |      84020 |     0 |    0 |     84020 |      316971 |

Don't know why the elf2bin is presenting the wrong numbers.

theotherjimmy, 2017-01-18:
@JanneKiiskila @andreaslarssonublox It sounds like this is a bug with memap.py. @MarceloSalazar Any thoughts on what could be causing this?

0xc0170, 2017-01-27:
> @JanneKiiskila @andreaslarssonublox It sounds like this is a bug with memap.py. @MarceloSalazar Any thoughts on what could be causing this?

@MarceloSalazar Any update?

MarceloSalazar, 2017-01-27:
I'm sorry, didn't have chance to look into this yet...

MarceloSalazar, 2017-05-03:
The patch file is indeed defined in the '.firmware' section, however this is not part of the GNU GCC standard sections, and hence why it's discarded.
https://gcc.gnu.org/onlinedocs/gccint/Sections.html

The problem doesn't happen when using the ARM or IAR compilers, because of the nature of the memory map files (this falls in the RO area).

@andreaslarssonublox would it be possible to rename .firmware to .text?

andreaslarssonublox, 2017-05-03:
Thanks for finding it @MarceloSalazar. We will fix it in the next release of the driver.

MarceloSalazar, 2017-05-19:
Built Wifi example with UBLOX_EVK_ODIN_W2 and mbed OS SHA: 5d0ce3c53158c69b5456d1d2ef4ecc12fdcfc49e

Total Flash memory (text + data + misc): **998507** bytes

Issue considered fixed.

closed 2017-05-19 by MarceloSalazar