ARMmbed/mbed-os #3606: Exporting to DS-5 will result in build error

TomoYamanaka, 2017-01-18:
### Description
- Type: Question
- Priority: Major 

---------------------------------------------------------------
## Question
When exporting the existing mbed classic program such as mbed_blinky to **DS-5**, the following error is shown in building the exported project:<br>

`mbed_blinky_ds5_5_rz_a1h\mbed_blinky\mbed/drivers/Serial.h:70:70: error: 'MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE' was not declared in this scope`

It seems to be due to lack of include path setting for `mbed_config.h`.
Therefore, it seems that `mbed_config.h` should be manually included at present so that we can build the project.

Just for information, this problem does not seem to occur until mbed lib revision 127.

Reference : https://github.com/ARMmbed/mbed-os/issues/3593

theotherjimmy, 2017-01-18:
Hey @TomoYamanaka You seem to be hitting all of the exporters that have not been updated in a bit. As a follow up to #3593 and this issue, I'll let you know the of the order that I will fix them in after I discuss it with @sg-.

TomoYamanaka, 2017-01-19:
Hi @theotherjimmy 

Thank you for your commnets.
I'm looking forward to hearing from you.

mgiaco, 2017-01-23:
Hello Is there some workaround or solution available in the near future because as I found out ARM GCC, STM System Workbench also did not work.

bridadan, 2017-01-26:
@mgiaco At the moment there isn't a set time frame for the DS-5 fix, but it is on our list of ToDos! In the meantime if you're just trying to use the ARM toolchain with an IDE, the Keil uVision exporter is working.

mgiaco, 2017-01-26:
@bridadan .. I do not need any fix for DS-5. I do not have that toolchain. That was a message from me for a general solution because I think almost evere GCC exporter does not work at the moment. The only thing I found and which I like most coming up is the gnuarmeclipse exporter which works really great.

theotherjimmy, 2017-01-26:
> I think almost every GCC exporter does not work at the moment.

Offline or online?

theotherjimmy, 2017-01-26:
And when you say "almost every" which ones are you talking about?

mgiaco, 2017-01-26:
sorry the part which does not work for me was online.  the gnuarmeclipse part i use is offline.

theotherjimmy, 2017-01-26:
> sorry the part which does not work for me was online.

Yeah, that's true at the moment.

> the gnuarmeclipse part i use is offline.

I really appreciate you testing that for us. I intend to use the knowledge gained from that PR to improve some of the other eclipse-based IDEs

bridadan, 2017-01-26:
To expand on @theotherjimmy's comment, we suggest that you use the exporters that are available offline at the moment. There are known issues with the online exporters and a fix is in the works. Sorry for the inconvenience!

sg-, 2017-02-07:
@theotherjimmy has the deployment fixed this?

theotherjimmy, 2017-02-07:
The original error is no longer seen. OTOH, the DS-5 exporter does not seem to take the scanned resources into account, so it tries to compile everything under the root but only specifies the include paths that are present in the scanned resources. This results in compiler errors.

peeyushpashine, 2017-06-23:
Is there any other workaround if I do not want to include mbed_config.h in RawSerial and Serial.h? For me, these are server files and I can not change them.

TomoYamanaka, 2017-07-25:
Hi, 

It seems that don't fix the issue.
I confirmed and used **mbed_blinky(rev 147)**, unfortunately I faced the above issue(not included `mbed_config.h`).
Any updates ?

TomoYamanaka, 2018-12-28:
I close this issue since DS-5 was removed from Mbed Exporter by #8590 .

closed 2018-12-28 by TomoYamanaka