ARMmbed/mbed-os #3609: Building with RTOS via build.py does not build the appropriate code

tim-nordell-nimbelink, 2017-01-18:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Expected behavior**
That source files such as "PlatformMutex.h" not compile in the stub implementations when the "--rtos" flag is provided to build.py, but would utilize the RTOS provided implementations.

**Actual behavior**
PlatforMutex.h (and others) use the stub implementation instead of the rtos implementation.  This is (in part) due to "MBED_CONF_RTOS_PRESENT" not being defined for the build of mbed.

**Steps to reproduce**
1. Checkout mbed-os repository
2. Edit "platform/PlatformMutex.h".  Insert an "#error" line in the stub build to demonstrate that the stub is being built.
3. Run "python tools/build.py -m NUCLEO_L476RG -t GCC_ARM --rtos -j1".  This will eventually fail with:

```
Compile [ 59.1%]: AnalogIn.cpp
[Error] PlatformMutex.h@26,0: #error 
In file included from /home/tim/idle/lib/mbed/BUILD/mbed/drivers/AnalogIn.h:25:0,
                 from /home/tim/idle/lib/mbed/drivers/AnalogIn.cpp:17:
/home/tim/idle/lib/mbed/BUILD/mbed/platform/PlatformMutex.h:26:2: error: #error 
```

0xc0170, 2017-01-19:
Thanks for reporting.

This is intentional, mbed 2 (scripts included like build.py) does not include thread safety. To have it in, please use mbed OS 5.

bulislaw, 2017-01-19:
I think it's confusing, obviously no one starts at the docs, so people get the sources have a look at the 'tools' and run `build.py` cause that sounds about right and it's before `compile.py` maybe we could isolate tools in `mbed2` `mbed5` folders?

bulislaw, 2017-01-19:
https://github.com/ARMmbed/mbed-os/pull/3613 Extends README to point people new to mbed OS to mbed-cli and documentation.

tim-nordell-nimbelink, 2017-01-19:
Is it expected that the wait API (which does not involve synchronization) also do busy waiting instead of using the RTOS features in this case?

Additionally, the scripts in "tools" are maintained with the repository - mbed-cli is not.  The repository should be able to build itself without special external tools.  Is there a way to build it with synchronization without using an external special tool and get a set of ".a" and ".o" files?

tim-nordell-nimbelink, 2017-01-19:
Previously (probably mbed os 2) an exported project from the online toolset to GNU Makefiles had used the python scripts (such as make.py), but now I see that the mbed-cli tool's export to GNU Makefile's is a bit more of a direct translation.  Digging a bit further, it looks like this invokes "tools/project.py" to generate this.  I'll make this work for my needs.  I'll leave this open until there is a response for "mbed_wait_api_no_rtos.c" vs "mbed_wait_api_rtos.cpp".  It still seems like it should not use the purely busy-wait API for even "mbed os 2".

0xc0170, 2017-01-27:
> Is it expected that the wait API (which does not involve synchronization) also do busy waiting instead of using the RTOS features in this case? mbed_wait_api_no_rtos.c and mbed_wait_api_rtos.cpp are also dependent on this variable being set.

Yes expected, how it used to be, and it is for mbed 2.

> Additionally, the scripts in "tools" are maintained with the repository - mbed-cli is not. The repository should be able to build itself without special external tools.

As you pinpointed in the last comment, mbed-cli sits on top of tools (uses them). I'll close this issue as it an answer was provided.

closed 2017-01-27 by 0xc0170