ARMmbed/mbed-os #3610: __real__calloc_r can clear too much memory (when running in uVisor)

janjongboom, 2017-01-18:
### Description
- Type: Bug
- Priority: Blocker for people who use uVisor with default settings

---------------------------------------------------------------

## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]

**mbed-cli version:** 
1.0.0

**meed-os sha:**
HEAD2885c1b41e63158cb6faf5f107cd821ae06ef26c

---

## Demo

* https://github.com/janjongboom/mbed-os-example-uvisor/tree/calloc_bug.
* Build with K64F and GCC_ARM.
* Issue shows in both default and debug profile.
* Issue shows both with and without gdb attached.

---

I realize this sounds insane, but `__real__calloc_r` - called from a uVisor box - can clear more memory than it allocates, thus causing all kinds of crazy errors. Ericsson approached us with memory issues when running the `mbedtls_ecp_mul` mbed TLS function within a uVisor box (but not in a normal thread), and I flew down to Sweden to investigate... Here's what I saw:

* There is plenty of space on the heap. ~10K free.
* The function can run in <2K (with `MBEDTLS_ECP_WINDOW_SIZE` configured) as seen on the main thread.
* Directly after calling this function the complete heap is full, every `malloc` call fails.

After a few hours of investigating we consulted @niklas-arm who suspected the memory allocator (uVisor uses a different allocator than the rest of mbed OS). This allocator is pretty simple, it allocates the full block of memory the uVisor box might use and then has a linked list of malloc'ed blocks. If a new block needs to be malloced it looks for a hole in memory that is big enough, and if so, allocates that block and updates the `next` pointer.

The last item of the linked list is an element at the end of the memory block (so there is a big hole between the semi-last and last block if your heap is not full nor fragmented). The last element [then points](https://github.com/ARMmbed/mbed-os/blob/04a31f3b39bfbd69e155ed7c9f55880fd75278ba/rtos/rtx/TARGET_CORTEX_M/rt_Memory.c#L86) to `NULL` to indicate the list ended. So the memory structure looks like (these are the actual addresses from the program linked below, compiled with GCC_ARM against K64F with debug profile):

```
0x1fff4508 -> ... -> 0x1fff5c48 -> 0x1fff5c68 -> ... -> 0x1fff94f8
```

[image]

*Last few items of the linked list*

Now... The 25th `calloc` of the program tries to allocate 32 bytes (incl. the overhead of the allocator this is 40 bytes). This call goes through rt_Memory.c correctly. The memory is allocated and the tree is updated. The pointer returned is `0x1fff5c48`, and the calloc'ed length is 8*4. This is then returned to rtx_malloc_wrapper.c where the pointer is correct.

When you inspect 0x1fff5c68 (when still in rtx_malloc_wrapper) it still contains the pointer to the next item in the list.

Now step through back to mbed_alloc_wrappers.cpp, right after the `__real__calloc_r` function returns. The `ptr` variable contains the right value, but if you inspect 0x1fff5c68 it is now cleared...

I've augmented the code like this:

```cpp
    printf("__real__calloc_r %p\r\n", ((MEMP*)0x1fff5c68)->next);
    // nmemb = 8, size = 4 here
    ptr = __real__calloc_r(r, nmemb, size);
    // ptr returned is 0x1fff5c48
    printf("__real__calloc_r_done %p\r\n", ((MEMP*)0x1fff5c68)->next);
```

This outputs the following:

```
__real__calloc_r 0x1fff5c90
__real__calloc_r_done 0x0
```

So `0x1fff5c68` is now cleared, pointing to 0x0. When we want to allocate more memory the allocator looks at the address, sees it points to 0, thinks it's the end of the list and refuses to allocate any more memory.

When you declare the `MBED_HEAP_STATS_ENABLED` macro the program uses `malloc`/`setmem` instead of the `__real__calloc_r` call and everything works 100%. No memory corruption.

You can see this issue in [this program](https://github.com/janjongboom/mbed-os-example-uvisor/tree/calloc_bug). Just build and run on your board. It will warn when the memory is going to be cleared (consistently shown on both my and Ericsson's board) with a `JUST BEFORE WE'RE GONNA FAIL` message on serial. Thereafter the allocation tree is corrupted and no more memory can be allocated. Declare the  `MBED_HEAP_STATS_ENABLED` macro and all will run as normal.

/cc @AlessandroA @meriac @sg- @yanesca

janjongboom, 2017-01-18:
Also easy to see this in action by attaching gdb and putting a watch at `*0x1fff5c90`. A write action originating from `__real__calloc_r` will happen, but with no stack trace...

niklas-arm, 2017-01-19:
Hm. [We never wrapped calloc_r for uVisor](https://github.com/ARMmbed/mbed-os/commit/7c0cc50677d17976d232b4b59a62e09d261d07f4), assuming newlib would only do malloc and memset, [which is not the case however](https://github.com/bminor/newlib/blob/master/newlib/libc/stdlib/mallocr.c#L3183-L3247).

Note that there may be more functions which need to be wrapped (memalign?, valloc?). newlib is fighting us here, since it was not designed to allow exchanging the dynamic memory implementation. 😒

moonchen, 2017-10-24:
I'm thinking of writing a patch that fixes this by moving the size metadata into a separate structure, indexed by the address of the allocation.  This would prevent memory corruption that occurs due to un-wrapped functions.  Un-wrapped functions would still result in potentially incorrect stats.  This fix is not directly targeted at uVisor, but rather the problem of unwrapped functions in general.

If you can think of problems with this approach, please let me know.

0xc0170, 2017-11-10:
@moonchen a patch welcome !

deepikabhavnani, 2018-09-11:
Will not be fixed: uVisor is deprecated since 5.9 and will be removed in 5.10 release. 
Reopen if issue exists in other parts of mbed-os.

closed 2018-09-11 by deepikabhavnani