ARMmbed/mbed-os #3622: ARM GCC toolchain not initializing RTOS

0x6d61726b, 2017-01-22:
Hi, I had some problems running mbed code compiled with ARM GCC toolchain. For details, see: [ARM GCC toolchain not initializing RTOS (was: simple RTOS example ends in HardFault_Handler)](https://developer.mbed.org/questions/76743/ARM-GCC-toolchain-not-initializing-RTOS-/).

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
LPC176X

**Toolchain:**
GCC_ARM

**mbed-os:**
mbed-os master (5.2.1)

**Expected behavior**
[RTOS Thread example](https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/APIs/tasks/rtos/) is running and the LEDs are flashing.

**Actual behavior**
HardFault_Handler() is called when the (RTOS) Thread is created. The problem is that the necessary RTOS initialization routines were not called.

**Steps to reproduce**
Compile the code with a GCC_ARM compiler using the [current startup file](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_ARM/startup_LPC17xx.S).

**Possible solution**
I created the following code (.c file instead of the .S file) which worked in that case and also for the [HTTP client program example](https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/APIs/communication/ethernet/). I have not tested any possible case the mbed-os library is capable of. 

`/*----------------------------------------------------------------------------`
`    Description: LPC17xx startup file for ARM GCC compilers`
`    Copyright:   unknown, files did not contain any copyright info`
`    License:     unknown, files did not contain any license info`
`- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -`
`    Note: This file is a adption of the basic startup code from Sysprogs`
`          (ORIGINAL_CR_startup_lpc175x_6x.c), extended with functionality from`
`          mbed TOOLCHAIN_GCC_CR (startup_LPC17xx.cpp) to call the necessary`
`          routines to run CMSIS RTOS/RTX from mbed.`
`- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -`
`    THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR`
`    IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,`
`    FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL`
`    THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER`
`    LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING`
`    FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER`
`    DEALINGS IN THE SOFTWARE.`
`-----------------------------------------------------------------------------*/`
` `
`// common defintions`
`#define NULL ((void *)0)`
` `
`// Local functions`
`void Reset_Handler();`
`void Default_Handler();`
` `
`// External functions`
`extern void SystemInit(void);`
`extern void pre_main(void) __attribute__((weak));`
`extern void software_init_hook(void);`
`extern void __libc_init_array(void);  // this comes from newlib-nano`
`extern int main(void);`
` `
`// The following external defintions are part of the linker script`
`extern void* __StackTop;`
`extern void* __etext;`
`extern void* __data_start__;`
`extern void* __data_end__;`
`extern void* __bss_start__;`
`extern void* __bss_end__;`
[... 117 lines cut ...]
`    SystemInit();`
`    `
`    if (pre_main)`
`    {`
`        // give control to the RTOS`
`        software_init_hook(); // this will also call __libc_init_array`
`    }`
`    else`
`    {`
`        // for BareMetal (non-RTOS) build`
`        __libc_init_array();`
`        (void)main();`
`    }`
`    `
`    // done - this should never be reached`
`    for (;;) ;`
`}`
` `
` `
`// The Default_Handler that gets executed if your code is missing an interrupt handler`
`void __attribute__((naked, noreturn)) Default_Handler()`
`{`
`    //If you get stuck here, your code is missing a handler for some interrupt.`
`    for (;;) ;`
`}`

bridadan, 2017-01-23:
@0x6d61726b Thanks for raising the issue! Can confirm that the RTOS example that you mentioned in your developer.mbed.org post is also broken in the latest release of mbed-os? The latest release is 5.3.3. I recently ran tests on the ARCH_PRO platform (it has an LPC1768 MCU) and non of the automated tests relating to the RTOS had any issues.

0x6d61726b, 2017-01-23:
Hi Brian, I can confirm that this issue still persists in the current mbed-os branch 5.3(.3).
The code runs fine when compiled with LPCXpresso (TOOLCHAIN_GCC_CR), but unfortunately not with TOOLCHAIN_GCC_ARM.

The issue is, that `ldr    r0, =_start` calls crt0.S which loads main(). The necessary initialization for RTOS `software_init_hook();` is never called, as done in the [startup code for TOOLCHAIN_GCC_CR](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.3/targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_CR/startup_LPC17xx.cpp#L155).

bridadan, 2017-01-24:
Thanks for following up @0x6d61726b, can you tell me what version of the GCC_ARM toolchain you're using?

0x6d61726b, 2017-01-24:
I am using VisualGDB (with SysGCC), that has GCC 4.9.2

0xc0170, 2017-01-25:
To understand, using GCC ARM or SysGCC ?

I dont recall seeing any problem for mbed lpc1768 with ARM GCC, it has been used for years.. As reference, the similar issue was reported for another platform here, https://github.com/ARMmbed/mbed-os/issues/775 (was fixed by calling ``_start``).

0x6d61726b, 2017-01-30:
Hi, it seems I had an issue with the toolchain. After setting it up again, the issue has disappeared. Sorry for any inconvenience caused.

closed 2017-01-30 by 0x6d61726b