ARMmbed/mbed-os #3623: Configure main thread stack size

ccli8, 2017-01-23:
### Description
- Question
- Minor
-----------------------------------------------------------------

## Question
How could I configure main thread stack size in *mbed_app.json* or other means?

bridadan, 2017-01-23:
cc @c1728p9 @0xc0170

0xc0170, 2017-01-24:
It is currently defined in ``RTX_CM_lib.h`` via ``DEFAULT_STACK_SIZE`` macro (that one comes from cmsis_os.h header file). This should be improved with new RTX coming.

cc @bulislaw

bridadan, 2017-01-26:
Is this being tracked elsewhere? If so, can we close this?

bulislaw, 2017-01-26:
Tracked in https://github.com/ARMmbed/mbed-os/issues/3315 closing this one.

closed 2017-01-26 by bulislaw

janjongboom, 2018-03-12:
Note that this is configurable in mbed_app.json now, see https://os.mbed.com/docs/v5.7/reference/configuration.html

Noting this down as this is the top result in Google.