ARMmbed/mbed-os #3625: Support small-memory chips

ccli8, 2017-01-23:
### Description
- Type: Question
- Priority: Minor
-----------------------------------------------------------------

## Question
We are evaluating adding mbed support for Cortex M0 MCUs. These MCUs have less SRAM, e.g. 16 KB. Some questions need your help to comment for these small-SRAM MCUs:
1. Per our evaluation, Greentea test will fail with ARM/GCC_ARM/IAR toolchain due to insufficient heap, but OK with uARM toolchain. Can it still declare as mbed OS 5 enabled?
1. As I know, mbed Online Compiler just supports ARM toolchain. Do you have plan to support uARM toolchain online?

cyliangtw, 2017-01-24:
Both of NUCLEO_F070RB and NUCLEO_F072RB targets are only 16KB SRAM, it still could be mbed OS 5 enabled. More supported_toolchains as ["ARM", "uARM", "IAR", "GCC_ARM"] in target.json .
@screamerbg  Mihail, how NUCLEO_F070RB could pass Greentea test with ARM/GCC_ARM/IAR toolchain ?

0xc0170, 2017-01-24:
> Per our evaluation, Greentea test will fail with ARM/GCC_ARM/IAR toolchain due to insufficient heap, but OK with uARM toolchain. Can it still declare as mbed OS 5 enabled?

Needs to support IAR, ARM and GCC_ARM - https://www.mbed.com/en/about-mbed/mbed-enabled/mbed-enabled-program-requirements/

> As I know, mbed Online Compiler just supports ARM toolchain. Do you have plan to support uARM toolchain online?

No, it is not.

>  @screamerbg Mihail, how could NUCLEO_F070RB pass Greentea test with ARM/GCC_ARM/IAR toolchain ?

cc @bcostm @LMESTM

ccli8, 2017-02-10:
Any update? Could 16-KB MCU still pass mbed OS 5 verification?

cyliangtw, 2017-04-06:
@screamerbg , according to greentea test's memory footprint, 16KB SRAM MCU can't afford mbed OS 5.x validation. Is there any plan to adjust the test criteria of mbed OS 5.x ?
Or, 16KB SRAM MCU only could apply mbed OS 2.0 ?

cyliangtw, 2017-04-07:
@screamerbg, @0xc0170 
I verified 2 mbed platforms with 16KB SRAM, one is ST-F302, the other is FRDM-K20D50M, both of two can’t pass mbed test. How about the validation criteria for this segment "16KB SRAM" ?

0xc0170, 2017-04-21:
cc @sg-

cyliangtw, 2017-04-21:
@0xc0170 
cc @sg- 
I will close this issue by next Monday.
Hopefully, CMSIS v5 and RTX v5 flexible main & child thread stack size seems meet 16KB segment.

ccli8, 2017-04-24:
The issue is expected to fix with CMSIS5/RTX2 which supports adjustment of main/child thread stack size.

closed 2017-04-24 by ccli8