ARMmbed/mbed-os #3633: [Exporter] [TLS-Client] : Platform not supported by IAR.

andresag01, 2017-01-24:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-tls/issues/34
- Priority: Minor

This was an issue originally files in the mbed-os-example-tls repo, but it is related to the exporter rather than the example code itself. Therefore, I am opening this issue and closing the original https://github.com/ARMmbed/mbed-os-example-tls/issues/34.

The problem is that during testing it was found that mbed-cli throws an error when trying to export a project to IAR embedded workbench for the NUMAKER_PFM_NUC472. Is this expected?

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
IAR

**Toolchain version:**
N/A

**mbed-cli version:** 
1.0.0

**meed-os sha:**
34c1fac Merge pull request #3585 from ARMmbed/release-candidate

**Expected behavior**
Export command should succeed.

**Actual behavior**
When trying to export the mbed TLS example to IAR the following error is printed:
```
--> mbed export -i IAR -m NUMAKER_PFM_NUC472
/usr/local/lib/python2.7/dist-packages/fuzzywuzzy/fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: NUMAKER_PFM_NUC472 not supported by iar
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u /home/andama01/Documents/mbed-os-example-tls-copy/authcrypt/mbed-os/tools/project.py -i iar -m NUMAKER_PFM_NUC472 --source ." in "/home/andama01/Documents/mbed-os-example-tls-copy/authcrypt"
```

**Steps to reproduce**
```
git clone https://github.com/ARMmbed/mbed-os-example-tls.git
cd authcrypt
mbed deploy
mbed export -i IAR -m NUMAKER_PFM_NUC472
```

bridadan, 2017-01-25:
At this time, the NUMAKER_PFM_NUC472 does not support exporting to IAR. Support is being added here: https://github.com/ARMmbed/mbed-os/pull/3590

theotherjimmy, 2017-01-25:
I'm closing this as it is expected behavior.

closed 2017-01-25 by theotherjimmy