ARMmbed/mbed-os #3634: Why pinmap_pinout is setting pin into PullNone mode ?

LMESTM, 2017-01-24:
I am wondering why the below line of code was introduced:
https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_pinmap_common.c#L27

This makes things more complex than should be, at least for STM targets, and maybe for others as well.

Here is what happens:

pinmap_pinout 
               |_ pin_function
                          |_ here the target specific code, selects an alternate function.
                              in case of STM32 targets, it also selects default mode (OpenDrain, PullDown ...)
               |_ pin_mode(pin, PullNone);
                           |_ here the common code over-writes the default configuration set above
this means the driver now specifically has to set mode again, calling to 
               |_ pin_mode(pin, PullDown );

I would consider removing the line, 
https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_pinmap_common.c#L27
assuming that targets have default mode settings that are possibly a better fit than PullNone.

Any feedback ?

bridadan, 2017-01-25:
@LMESTM looks like that line has existed since practically the beginning of time :smile: 

Any thoughts @0xc0170

LMESTM, 2017-01-25:
@bridadan - yes this was part of the initial commit as far as I've seen - so I couldn't find out any comment about it. But I still consider that it may be removed ...

0xc0170, 2017-01-25:
``pin_function`` does not specify what it should/not do (one of the docs we need to write!). Therefore with this assumption, a function ``pinmap_pinout`` sets it to default one.  If we remove it now, before the docs, we should have a look at all pin_functions what they do, as it can break the software (generic layer is setting it to PullNone.

>  assuming that targets have default mode settings that are possibly a better fit than PullNone.

what mode it should be then by default?

> This makes things more complex than should be, at least for STM targets, and maybe for others as well.

Is there anything more than the call tree you shared?

LMESTM, 2017-01-25:
@0xc0170 
> what mode it should be then by default?
I think this is up to targets (sw or hw) to set the default mode, not up to this function. 
>Is there anything more than the call tree you shared?
Not more. But that applies to all drivers and is misleading. 

Another possible use case: a user may want to share a pin between 2 IPs and use PullUp to maintain the pin state during transition from one IP to another. The transition from one IP to another is done with a call to pin_function, which would overwrite the PullUp mode with NoPull and may create undesired pin state.

closed 2017-10-27 by ChiefBureaucraticOfficer