ARMmbed/mbed-os #3636: mbed TLS examples compilation failing in online IDE

andresag01, 2017-01-24:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-tls/issues/32
- Priority: Major

Compiling the TLS client example (from https://github.com/ARMmbed/mbed-os-example-tls) using the Online IDE and K64F results in a failure. Note that this issue was originally raised in the mbed TLS examples repository (https://github.com/ARMmbed/mbed-os-example-tls/issues/32). However, the problem cannot be reproduced using the command line tools (mbed-cli) or the Makefiles (using `mbed export`).

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
Online IDE at https://developer.mbed.org/compiler/. I suspect it uses ARMC5

**meed-os sha:**
#34c1facf42a174f47fdf9002cd8c6bf10ac41744 (The default for mbed TLS examples)

**Expected behavior**
The application should compile without errors.

**Actual behavior**
The compilation process results in the following error:
```
Error: Cannot open source input file "device.h": No such file or directory in "extras/tls-client/mbed-os.lib/platform/platform.h", Line: 27, Col: 21
```

**Steps to reproduce**
* Import the mbed TLS examples from the URL https://github.com/ARMmbed/mbed-os-example-tls.git into the online IDE.
* Change the platform to K64F
* Compile the tls-client application
* Observe the failure message at the end of the compilation process.

bridadan, 2017-01-26:
The GitHub repository you posted contains multiple examples. This is not supported by the Online IDE. When you import a repository as a program, it is assumed that there is only one project present in the repo.

When compiling TLS examples in the Online IDE, users should use the examples mirrored to the developer site:

- https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-benchmark
- https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-tls-client
- https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-hashing
- https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-authcrypt

@andresag01 If you wish to continue having multiple examples in the GitHub repository you mentioned, it may be worth linking to the developer site mirrors for use with the Online IDE in your README. What do you think?

andresag01, 2017-01-26:
@bridadan: Thanks for the information. We will just add links to the mirrors to our documentation.

bridadan, 2017-01-27:
Thanks, I'll go ahead and close this

closed 2017-01-27 by bridadan