ARMmbed/mbed-os #3637: mbed TLS benchmark example fails compilation in online IDE for NUC472

andresag01, 2017-01-24:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-tls/issues/33
- Priority: Major

This is an issue that was originally reported in the mbed TLS examples repo at https://github.com/ARMmbed/mbed-os-example-tls/issues/33. The problem is that compilation of the mbed TLS benchmark example fails for the platform NUMAKER_PFM_NUC472 when using the Online IDE. However, it does work correctly when compiling using the command line tools (mbed-cli) or the Makefiles generated by the exporter (mbed export).

I did some brief investigation and it seems that not all the macros in the `device_has` field of the target.json for the NUMAKER_PFM_NUC472 are present when compiling with the online IDE. Specifically, I found out that the macro `DEVICE_TRNG` is not present. This macro enables usage of entropy sources for mbed TLS in mbed OS, this might be part of the problem.

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
I do not know what is used in the Online IDE at https://developer.mbed.org/compiler/. I suspect is ARMC5.

**mbed-os sha:**
#34c1facf42a174f47fdf9002cd8c6bf10ac41744. The default in the example's `mbed-os.lib` file.

**Expected behavior**
The example should compile without errors.

**Actual behavior**
When compiling with the online IDE, the following compilation errors are reported:
```
Error: Undefined symbol mbedtls_hardware_poll (referred from entropy.NUMAKER_PFM_NUC472.o)
```

**Steps to reproduce**
* Import the mbed TLS examples from the URL https://github.com/ARMmbed/mbed-os-example-tls.git into the online IDE.
* Change the platform to NUMAKER_PFM_NUC472.
* Compile the benchmark application
* Observe the failure message at the end of the compilation process.

bridadan, 2017-01-26:
@andresag01 I can't reproduce this issue when using the developer site mirror of the benchmark example here: https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-benchmark/

Can you confirm this as well? If you find this works, please see my comment here: https://github.com/ARMmbed/mbed-os/issues/3636#issuecomment-275382449

Thanks!

andresag01, 2017-01-26:
@bridadan: As you said, the problem seems to be that the Online IDE does not support multiple examples under the same project directory. I will add links to the mirrors to mbed TLS example's documentation.

bridadan, 2017-01-27:
Thanks @andresag01! Closing this now.

closed 2017-01-27 by bridadan