ARMmbed/mbed-os #3638: Where are all the exporters in the online compiler?

janjongboom, 2017-01-24:
From CLI I can export to way more platforms than in the online compiler, f.e. Eclipse is missing for K64F. What's going on here?!

@sg- @screamerbg @theotherjimmy

theotherjimmy, 2017-01-25:
>What's going on here?!

The tools used on the online compiler are out of date. We are aware and working to close the gap between the mbed-os tools and the online compiler's version.

janjongboom, 2017-01-25:
@theotherjimmy Any timeframe on that? I'm updating the debugger docs and would like the Eclipse exporter.

theotherjimmy, 2017-01-25:
No time frame yet. @immunda any comments?

immunda, 2017-01-25:
We're redeploying some large changes to the site today. Provided that's all stable through tomorrow I'll then merge in these tools and deploy then.

sg-, 2017-02-07:
@theotherjimmy @immunda Has this been deployed?

janjongboom, 2017-02-07:
I see new exporters now, although they have no friendly names (just called `eclipse_gcc_arm` and they don't have logos)... @screamerbg

theotherjimmy, 2017-04-25:
@janjongboom Can we close this issue? The exporters are there.

closed 2017-05-01 by janjongboom