ARMmbed/mbed-os #3639: Export broken in the online compiler

janjongboom, 2017-01-24:
* Blinky on latest mbed OS 5
* Export to K64F -> GCC (ARM embedded)

[image]

@screamerbg @theotherjimmy

theotherjimmy, 2017-01-25:
Thanks @janjongboom. We are aware of this issue (I get an email for all website tracebacks!) and are working to update the tools used on the website to what we are releasing with mbed-os.

sg-, 2017-02-07:
@theotherjimmy has this been resolved?

theotherjimmy, 2017-02-07:
I have been able to export that project recently to that target and toolchain combination. I think it has been resolved.

closed 2017-02-07 by sg-