ARMmbed/mbed-os #3643: mbed TLS examples fail intermitently when using the online IDE

andresag01, 2017-01-25:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os-example-tls/issues/31
- Priority: Minor

This issue was originally reported in the mbed TLS examples repository: https://github.com/ARMmbed/mbed-os-example-tls/issues/31.

The mbed TLS examples intermittently fail to compile when using the mbed online IDE. The failure causes vary between the examples and sometimes the same example exhibits different faults depending on the time of compilation. Errors are related to:

* The online IDE fails to load, hangs, seems to be stuck doing some operations.
* Compilation errors are reported for no good reason since the programs compile without problems using the offline tools e.g. mble-cli, Makefiles, IAR, uVision, etc...

---------------------------------------------------------------
## Bug

**Target**
Its irrelevant, the same issue happened when compiling for multiple targets. (e.g. K64F, NUMAKER_PFM_NUC472, etc)

**Toolchain:**
The compiler used by the mbed online IDE at https://developer.mbed.org/compiler/. I suspect its ARMC5.

**mbed-os sha:**
The default version when importing the following mbed TLS examples into the online IDE through the website:
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-tls-client/
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-benchmark/
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-hashing/
* https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-authcrypt/

**Expected behavior**
Compilation should succeed without errors.

**Actual behavior**
Compilation reports (different) errors intermittently.

**Steps to reproduce**
* Import an mbed TLS example into the IDE.
* Compile the program for one of the supported targets (say K64F).
* A compilation error is reported (note that it is an intermittent fault and we have no information regarding the probability of failure).

bridadan, 2017-01-26:
It sounds like this is an issue with the online IDE specifically, which unfortunately we have no control over here. I have seen intermittent issues as well (ex. `build system failed to finish successfully`). I don't really know what else to do about it though since that falls under @theotherjimmy and @screamerbg I believe. @theotherjimmy are these intermittent failures tracked somewhere? If so I'd like to close this issue.

sg-, 2017-02-07:
@theotherjimmy any ideas here? If infrastructure lets just close.

theotherjimmy, 2017-02-07:
The intermittent issues can be seen in the website's Sentry, which is always open on my browser. :sweat_smile:

closed 2017-10-27 by ChiefBureaucraticOfficer