ARMmbed/mbed-os #3645: Broken:mbed-os-example-client W/ UBLOX_EVK_ODIN_W2

andcor02, 2017-01-25:
Using mbed-os-example-client with UBLOX_EVK_ODIN_W2 

Example: https://github.com/ARMmbed/mbed-os-example-client

Importing example and switching to target UBLOX_EVK_ODIN_W2 as Ethernet or WiFi, compile all fine, flashes fine.

**ISSUE**

1. Board just hangs, outputs "Connecting to ethernet..." but nothing else happens?

2. When using WiFi, connects to network and registers, however after one button press on 'SW1' again device hangs.

0xc0170, 2017-01-25:
cc @andreaslarssonublox

andreaslarssonublox, 2017-01-26:
Can you have a look at this @andreaspeterssonublox?

andreaspeterssonublox, 2017-04-19:
Tested with ARMmbed/mbed-os-example-client release "mbed-os-5.4.3" and the WiFi and Ethernet applications connect, register and update the simulated button click every 15 seconds.

Before building the example I did the following according to the README.md:
- set up a .mbedignore file, an example file is available in the configs folder
- updated the security.h with content from mbed Device Connector > My Devices > Security credentials

Building the application was done with the -c --app-config configs\wifi_odin_v4.json and configs\eth_v4.json flags respectively.

Can you please re-test with the latest release (mbed-os-5.4.3) or later.

cmonr, 2018-06-14:
Closing since issue appears to be resolved.

Feel free to open a new issue if this is not the case.

closed 2018-06-14 by cmonr