ARMmbed/mbed-os #3646: uVision 5 export is broken - device.h missing

janjongboom, 2017-01-25:
1. Take mbed OS Blinky (K64F) in the online compiler
2. Export to uVision 5
3. Open the project
4. Build

Result: build fails.

```
mbed-os/./platform/platform.h(27): error:  #5: cannot open source input file "device.h": No such file or directory
  #include "device.h"
mbed-os/features/frameworks/utest/mbed-utest-shim.cpp: 0 warnings, 1 error
".\build\uvision5\mbed-os-example-blinky-serial.axf" - 62 Error(s), 1 Warning(s).
```

@theotherjimmy

0xc0170, 2017-01-27:
> @theotherjimmy

bump

bridadan, 2017-01-27:
I believe this issue is caused because the tools on the online compiler are out of date, which has been mentioned in multiple issues. Work is on going with the web team to fix this. In the meantime you should still be able to use mbed CLI to import the project and export it locally.

janjongboom, 2017-02-06:
@theotherjimmy @immunda  I thought this was fixed with a tools release online? Same problem still happens.

sg-, 2017-02-07:
@theotherjimmy bump

theotherjimmy, 2017-04-26:
I think this was the "scan for targets.json" issue we had a while back. It would not have picked up the right files because the Freescale label got changed to MCUXpresso. I should have resolved that issue a while back. Could we test/close this issue?

cmonr, 2018-01-23:
Confirmed build works on personal setup (after installing CMSIS Pack). Closing.

closed 2018-01-23 by cmonr