ARMmbed/mbed-os #3647: Request for mbed DMA driver for STM32

fwndz, 2017-01-25:
hi, 

Is there any mbed library that provides DMA library for STM32. I am try to run ADC through DMA. How can I implement it? Should I just straight away use the STM32 Hal? I think this way is so not practical
.If I want to create my own library can anyone briefly told me the architecture layer of mbed or provide the guideline  or code example.

Thanks

Regards
Fwndz

LMESTM, 2017-01-25:
Hi,
There is no generic MBED layer so far to configure DMA. So you need call to HAL layer for now. 
If you want to make it more generic and propose a library, you may add a new ADC interface in MBED layer (AnalogIn.h and hal/analog_in.h) that would request a number of samples for instance instead of a single sample, and then propose an implementation for STM32 targets (analogin_api.c).
ARM team would review the MBED layer proposal. ST team would review the STM32 implementation. This can then be integrated in MBED or used as a separate library.
Would that be ok ? 
Cheers
Laurent

gte1, 2017-01-25:
DMA would also be useful agains STM32 DAC / AnalogOut (audio/waveform playback etc.)

fwndz, 2017-01-26:
hi

Thanks for the prompt reply @LMESTM  and @gte1. Yes exactly, I am intend to implement continuous data sampling. Can I know which kind of microcontroller that already has DMA driver from mbed?

Cheers
Fwndz

LMESTM, 2017-01-26:
I don't think there existing examples to use for now.
As mentioned earlier, I would suggest to add a AnalogIn/AnalogOut API that provides the requires service (not specifically mentioning DMA). Do you have an idea of the functionality that your application need ? 
Then you can implement the service in the _api layer fo STM32 ralying on DMA usage with STM32 cube HAL API.

fwndz, 2017-01-26:
Yes I used to work with ADC through DMA before. All the HAL I configured manually. I notice we can implement the configuration manually in analogIn_api to enable the DMA. However I have one question regarding interrupt. How to implement it on mbed? 
```
void DMA2_Stream0_IRQHandler(void)
{
	  /** Test on DMA Stream Transfer Complete interrupt */
	  if(DMA_GetITStatus(DMA2_Stream0, DMA_IT_TCIF0))
	  {
	    ///< Clear DMA Stream Transfer Complete interrupt pending bit */
	    DMA_ClearITPendingBit(DMA2_Stream0, DMA_IT_TCIF0);

	    if(DMA_GetCurrentMemoryTarget(DMA2_Stream0)==1) ///< When DMA is targeting memory 1 (for receiving data)
	    {
	    	CalcADC_DMA_ReceiveBuffer(0); ///< Process/ calculate the received buffer in memory 0
	    }
	    else ///< When DMA is targeting memory 0
	    {
	        CalcADC_DMA_ReceiveBuffer(1); ///< Process/ calculate the received buffer in memory 1
	    }
	  
	  }

	  DMA_ClearITPendingBit(DMA2_Stream0, DMA_IT_HTIF0); ///< clear pending interrupt
}
```

I used to run my interrupt function as above. However, how can  I translate it into mbed. I know the process will be long but can you @LMESTM give me some guideline or hints on how interrupt can be used in my case.

Cheers

LMESTM, 2017-01-26:
@fwndz 

Just for information, which board do you plan to use ? 

In any case, the above code should work fine inside MBED, because DMA2_Stream0_IRQHandler is the name of the default IRQ as defined in the vector table, for instance here
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_NUCLEO_F411RE/device/TOOLCHAIN_GCC_ARM/startup_stm32f411xe.S#L207

Note: Sometimes in MBED a default IRQ handler might be replaced at runtime by another MBED handler,which can be done by a call to NVIC_SetVector like here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/i2c_api.c#L133

But for DMA, this is not the case, so the default handler is valid.
So feel free to use this piece code for your implementation

fwndz, 2017-01-26:
@LMESTM I am planning to use NUCLEO F302R8 Board for this application. Is there any issues regarding the DMA IRQ for this board?

LMESTM, 2017-01-26:
You just need to use the proper IRQ name for you board, as defined in the vector table (or to set your own one with NVIC_SetVector).

For NUCLEO_F302R8, the default DMA IRQ handlers are defined here
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F3/TARGET_NUCLEO_F302R8/device/TOOLCHAIN_GCC_ARM/startup_stm32f302x8.S#L299
So you need to pick-up to proper one : please have a look into the reference manual to find out which one

Note that we are planning to add some DMA support in MBED for STM32, but this will take some more time.

fwndz, 2017-01-26:
Thanks a lot for your support @LMESTM.

closed 2017-10-27 by ChiefBureaucraticOfficer