ARMmbed/mbed-os #3650: Can Bus IRQ_RX issue with infinity loop

khurss, 2017-01-26:
When i attach callback to can object and when i received message nuncleo go to infinite loop .

#include <mbed.h>

DigitalOut g_LED(LED1);
bool isMsgAvailable = false;
Serial pc(USBTX, USBRX, 921000);

void OnRecieve()
{
	isMsgAvailable = true;
}

int main() 
{
	CAN* _can = new CAN(PA_11,PA_12);
	_can->frequency(500000);
	_can->attach(&OnRecieve);
	
	CANMessage msg;
	
	for (;;)
	{
		if (isMsgAvailable)
		{
			isMsgAvailable = false;
			
			const auto is_ok = _can->read(msg); // can::read never called with rx interrupt
			if (is_ok) 
			{
				pc.printf("Can id : %d | len : &d \n\r", msg.id, msg.len);	
			}
		}

	}
}

0xc0170, 2017-01-27:
What target are you using  ? Be more specific please.

cc @bcostm @adustm @LMESTM @jeromecoutant

khurss, 2017-01-27:
Target is Nucleo F446RE

khurss, 2017-01-27:
Also function can->read should return 0 with no messages on can bus , 1 if message available. 
But in fact this function always return 1.  (can_api.c)

```
int can_read(can_t *obj, CAN_Message *msg, int handle) 
{
    //handle is the FIFO number

    CAN_TypeDef *can = (CAN_TypeDef *)(obj->can);    
    
    /* Get the Id */
    msg->format = (CANFormat)((uint8_t)0x04 & can->sFIFOMailBox[handle].RIR);
    if (!msg->format) {
        msg->id = (uint32_t)0x000007FF & (can->sFIFOMailBox[handle].RIR >> 21);
    } else {
        msg->id = (uint32_t)0x1FFFFFFF & (can->sFIFOMailBox[handle].RIR >> 3);
    }
      
    msg->type = (CANType)((uint8_t)0x02 & can->sFIFOMailBox[handle].RIR);
    /* Get the DLC */
    msg->len = (uint8_t)0x0F & can->sFIFOMailBox[handle].RDTR;
//  /* Get the FMI */
//  msg->FMI = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDTR >> 8);
    /* Get the data field */
    msg->data[0] = (uint8_t)0xFF & can->sFIFOMailBox[handle].RDLR;
    msg->data[1] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDLR >> 8);
    msg->data[2] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDLR >> 16);
    msg->data[3] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDLR >> 24);
    msg->data[4] = (uint8_t)0xFF & can->sFIFOMailBox[handle].RDHR;
    msg->data[5] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDHR >> 8);
    msg->data[6] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDHR >> 16);
    msg->data[7] = (uint8_t)0xFF & (can->sFIFOMailBox[handle].RDHR >> 24);
    
    /* Release the FIFO */
    if(handle == CAN_FIFO0) {
        /* Release FIFO0 */
        can->RF0R = CAN_RF0R_RFOM0;
    } else { /* FIFONumber == CAN_FIFO1 */
      /* Release FIFO1 */
      can->RF1R = CAN_RF1R_RFOM1;
    }

    return 1;
}
```

bcostm, 2017-02-01:
> Also function can->read should return 0 when message unavailable on can bus , 1 - if message available.
> But in fact this function always return 1. (can_api.c)

This has been discussed here Issue #3463 and normally fixed in PR #3492

sg-, 2017-02-07:
#3492 is merged. What else is needed to close this issue?

adustm, 2017-05-05:
bump ? @khurss Is it ok to close ?