ARMmbed/mbed-os #3656: I2C bus does not recover if RESET occurs within read

SeppoTakalo, 2017-01-27:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Expected behavior**
* You should expect I2C bus to be able to recover from reset condition.

**Actual behavior**
* If device reset within I2C read/write, there is no mechanism to recover. You need to unpower the device to recover.

**Steps to reproduce**

1. Create a loop that continuously reads any I2C devices register.
1. Press RESET button while read is ongoing.
1. If timing is correct, all subsequent read requests to device will fail. RESET does not help anymore.

**Example code**
Following code produces this problem using Atmel RF shield, by reading is AT24MAC register.

```
#include "mbed.h"
#include "NanostackRfPhyAtmel.h"
#include "at24mac.h"

Serial output(USBTX, USBRX);

int main(void) {

    output.baud(115200);
    printf("Starting\r\n");

    AT24Mac at24mac(ATMEL_I2C_SDA, ATMEL_I2C_SCL);
    uint8_t eui64[8];

    for (int i=0;;i++) {
        if (at24mac.read_eui64(eui64) == 0) {
            printf("Read eui64(%d) %hhx:%hhx:%hhx:%hhx:%hhx:%hhx:%hhx:%hhx\r\n",i, eui64[0],eui64[1],eui64[2],eui64[3],eui64[4],eui64[5],eui64[6],eui64[7]);
        } else {
            printf("Failed! %d\r\n", i);
        }
        Thread::wait(10);
    }
}
```

When all is working, this just outputs `Read ui64...` and the address it just  read.
Click reset for multiple times, until you start seeing `Failed!` messages.
From this condition, there is no way to recover.

There is internal ARM ticket open ONME-2963

0xc0170, 2017-01-30:
cc @mmahadevan108

SeppoTakalo, 2017-06-06:
There is a hacky workaround for this: https://github.com/ARMmbed/atmel-rf-driver/pull/54

So in theory, should be able to be fixed in mbed OS I2C API level.

0xc0170, 2017-06-06:
Thanks for sharing.

@SeppoTakalo Is this issue related (there are few suggestions for i2c reset): https://github.com/ARMmbed/mbed-os/issues/4220

0xc0170, 2017-06-06:
cc @chrissnow

SeppoTakalo, 2017-06-06:
Yes, seems to be related to this one.
At least the proposed fixes are the ones we applied in the RF driver.

SeppoTakalo, 2017-06-06:
Hey, this is actually not NXP specific. I believe it can happen on any platform.
All that is required is an external I2C device without reset pin.

chrissnow, 2017-06-23:
I've just implemented the fix from the Atmel driver referenced above and it seems to work.

is there any reason that this couldn't be added to the constructor of I2C?
Perhaps prior to `i2c_init(&_i2c, sda, scl);` ?
```c++
    mbed::DigitalInOut SDA(sda, PIN_OUTPUT, PullUp, 1);
    mbed::DigitalInOut SCL(scl, PIN_OUTPUT, PullUp, 0);
    //generate 9 clocks for worst-case scenario
    for (int i = 0; i < 10; ++i) {
        SCL = 1;
        wait_us(5);
        SCL = 0;
        wait_us(5);
    }
    //generate a STOP condition
    SDA = 0;
    wait_us(5);
    SCL = 1;
    wait_us(5);
    SDA = 1;
    wait_us(5);
```

this way the interface will get cleared up prior to use, then a generic i2c reset function could simply call the constructor again.

deepikabhavnani, 2017-07-31:
@SeppoTakalo - Will it be possible for you to test this fix #4838

SeppoTakalo, 2017-08-02:
@KariHaapalehto Can you check?

cmonr, 2018-10-08:
@KariHaapalehto also mentioend that https://github.com/ARMmbed/mbed-os/pull/8109 should fix the issue. Would help to see if this is confirmed.