ARMmbed/mbed-os #3659: Nordic targets fail to compile with macro `DEBUG` set

bridadan, 2017-01-28:
### Description
- Type: Bug
- Priority: Major

This bug was discovered when testing PR #3561 to add GNU ARM Eclipse exporter support. This may end up being a blocker because the Eclipse adds the `DEBUG` macro to the Debug builds (the default profile).

---------------------------------------------------------------
## Bug

**Target**
Nordic platforms (NRF51_DK was tested)

**Toolchain:**
All toolchains

**mbed-cli version:** 
1.0.0

**meed-os sha:**
15cd3da4ba009dbfc531df734029c0b74742a1e5

**Expected behavior**
Should compile

**Actual behavior**
Fails to compile with the error `fatal error: bsp.h: No such file or directory`

**Steps to reproduce**
```
$ mbed test -m NRF51_DK -t GCC_ARM --compile --clean -D DEBUG -v
...
        [DEBUG] Return: 1
        [DEBUG] Output: C:\Users\bridan01\Documents\dev\m_mbed\mbed\targets\TARGET_NORDIC\TARGET_NRF5\sdk\libraries\util\app_error.c:57:17: fatal error: bsp.h: No such file or directory
        [DEBUG] Output:  #include "bsp.h"
        [DEBUG] Output:                  ^
        [DEBUG] Output: compilation terminated.
```

The `bsp.h` file is available from the nordic SDK, however it includes other files as well. I was able to get it compiling by removing some of the includes, however I'm not sure if I removed anything important. I'd like to here from people who are more familiar with the nordic SDK before submitting a change.

cc @pan-

pan-, 2017-01-30:
@bridadan I'm under the impression that this issue is more related to the GNU ARM eclipse exporter than the Nordic code itself. I'm not sure it is a sane behavior to allow exporters to define macros not present in the initial project; especially if one of the goal of mbed cli is reproducibility. 

@nvlsianpu Do you think it is achievable to compile the SDK with `DEBUG` defined ? What is the impact on maintenance on your side ?

nvlsianpu, 2017-01-30:
Yes it's possible - it's one time job for few hour.

0xc0170, 2017-01-30:
> @bridadan I'm under the impression that this issue is more related to the GNU ARM eclipse exporter than the Nordic code itself. I'm not sure it is a sane behavior to allow exporters to define macros not present in the initial project; especially if one of the goal of mbed cli is reproducibility.

I wonder also why ``DEBUG`` is being added, I browsed few comments in the PR but could not find the reference there. If this is not common for all exporters, it shall not be defined. I can even find it in the files there.. Brian please can you provide more details?

This however looks like a bug as one macro should not break the build or ? How is that macro used within the SDK?

pan-, 2017-01-30:
> This however looks like a bug as one macro should not break the build or ? How is that macro used within the SDK?

It also looks like a *bug* to me and it should be addressed but it is orthogonal to the exporter GNU ARM eclipse exporter issue and I believe both issues should be acknowledge and treated respectively. The broken build is a side effect of the GNU ARM eclipse exporter inconsistency. 

@nvlsianpu Might be good to address this defect then. We can discuss about it tomorrow.

bridadan, 2017-01-30:
>  If this is not common for all exporters, it shall not be defined.

Totally agree, I think the behavior is not intended and should be unified with the other exporters (if possible). I only mentioned that PR because that's how discovered the issue.

> This however looks like a bug as one macro should not break the build

This was the main reason why I raised this issue :smile: `DEBUG` I imagine is a fairly common macro someone would add, I would hope we'd be able to cope with that.

nvlsianpu, 2017-01-30:
I agree - DEBUG issue is a bug. Probably it's consequence of that ARM compiler expects that all code in compiled sources are "not corrupted" even if that code ain't need for build.

smarek, 2017-03-31:
Trying to debug NRF51_DK project, I've overcome this issue by modifying Makefile like this (removing `-DDEBUG` from CC_FLAGS, adding `-g3` to CPPC_FLAGS)

Still would be better to have cleaner solution ;)

```diff
 ifeq ($(DEBUG), 1)
-  CC_FLAGS += -DDEBUG -O0
+  CC_FLAGS += -O0
+  CPPC_FLAGS += -g3
 else
   CC_FLAGS += -DNDEBUG -Os
 endif
```

jrobeson, 2017-09-16:
is this still valid anymore?

bridadan, 2017-09-19:
I just tried it with the latest master (003dd7c47f18628b10cddb3a81b572fa20a01032) and yup, same error:

```
$ mbed test -m NRF51_DK -t GCC_ARM --compile --clean -D DEBUG -v

[... SNIP ...]

[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -DTARGET_CORTEX_M -DTARGET_SDK11 -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DDEVICE_ANALOGIN=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DTARGET_MCU_NRF51_UNIFIED -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DDEBUG -DFEATURE_BLE=1 -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DNO_SYSTICK -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DMBED_BUILD_TIMESTAMP=1505829235.92 -DDEVICE_PWMOUT=1 -DS130 -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K @C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\BUILD\tests\NRF51_DK\GCC_ARM\.includes_1f92befe8c80e6e3894745e57bbce635.txt -include C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\BUILD\tests\NRF51_DK\GCC_ARM\mbed_config.h -MD -MF C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\BUILD\tests\NRF51_DK\GCC_ARM\targets\TARGET_NORDIC\TARGET_NRF5\TARGET_SDK11\libraries\util\app_error.d -o C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\BUILD\tests\NRF51_DK\GCC_ARM\targets\TARGET_NORDIC\TARGET_NRF5\TARGET_SDK11\libraries\util\app_error.o C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\targets\TARGET_NORDIC\TARGET_NRF5\TARGET_SDK11\libraries\util\app_error.c
        [Fatal Error] app_error.c@57,17: bsp.h: No such file or directory
        [DEBUG] Return: 1
        [DEBUG] Output: C:\Users\bridan01\onedrive_arm\Documents\dev\m_mbed\mbed\targets\TARGET_NORDIC\TARGET_NRF5\TARGET_SDK11\libraries\util\app_error.c:57:17: fatal error: bsp.h: No such file or directory
        [DEBUG] Output:  #include "bsp.h"
        [DEBUG] Output:                  ^
        [DEBUG] Output: compilation terminated.
```

cmonr, 2018-06-14:
Closing since issue should be resolved now, and last update is really old. Please open a new issue if the problem is still present.

closed 2018-06-14 by cmonr

bridadan, 2018-06-14:
The compilation still fails, but I'm not sure if we have to raise as an issue again, what do you think?

cmonr, 2018-06-15:
I've asked @pan- for guidance.