ARMmbed/mbed-os #3662: mbed storage: FATFileSystem::format() has incorrect allocation unit arg to f_mkfs()

simonqhughes, 2017-01-31:
There are 2 points to this issue:
- 	The FATFileSystem::format() calls f_mkfs with incorrect third argument if the sdcard has not been formatted with 512 byte cluster size. This is a coding bug and should be fixed.
-	How to expose f_mkfs() so this can be fixed in a general way.

## Bug

FATFileSystem::format() calls f_mkfs() with (likely) incorrect value for the allocation unit. This needs to be fixed

**Actual behavior**
See email below.

**Steps to reproduce**
See email below.

-----------------------------------------------------------------

email from Daniel Benor describing this issue:

From: Daniel Benor 
Sent: 30 January 2017 17:59
To: Simon Hughes
Cc: Clive Bluston
Subject: file system corruption investigation

hi Simon
Hope you feel better.
I wanted to share with you some findings related to file system corruption:
During development Clive found that the file system got corrupted once or twice a day.
I could reproduce the corruption by pressing the reset button during our tests. after 5-15 times I got a corruption that could be fixed with chkdsk in the PC.
I tried calling   f_mkfs(_fsid, 1U, 0U); in a small test after the corruption and I saw that this fixed the sd card, and there was no need to extract it and run chkdsk.

I tried calling sd.format from our CPP file that calls mbedOS API, but I got stuck. I changed the parameters of  f_mkfs as shown below and this solved the problem.
I added a call to sd.format on starting our tests and now I can't reproduce the file system corruption.
Even after corruption if I run our tests, sd.format is fixing the corruption.

The question is what kind of API you can expose to allow calling format. This will be very required in CI testing system.
Any comments or questions will be appreciated
Danny

int FATFileSystem::format() {
    lock();
    FRESULT res = f_mkfs(_fsid, 1U, 0U); // Logical drive number, Partitioning rule, Allocation unit size (bytes per cluster)
//    FRESULT res = f_mkfs(_fsid, 0, 512); // Logical drive number, Partitioning rule, Allocation unit size (bytes per cluster)
    if (res) {
        debug_if(FFS_DBG, "f_mkfs() failed: %d\n", res);
        unlock();
        return -1;
    }
    unlock();
    return 0;
}

dannybenor, 2017-01-31:
Hi
My question is more naive: Can we actually call sd.format from PAL code? I am asking because all our API is done via stdio.h and I am not sure if there is a difference in the call flow in terms of threads or similar.

simonqhughes, 2017-01-31:
@dannybenor Yes, you can call from PAL. Clive is OK with this.

sg-, 2017-02-07:
Why would PAL call sd.anything? That breaks portability given I could create a sd instance called sd_card or ssdd etc...

simonqhughes, 2017-02-08:
@sg- PAL team need sd.format() to recover inconsistent file systems back to a known state. This has been a problem for them during development and testing, and was requested prior to release to Jenia Kogan's team.

This has now been fixed with #3704.

closed 2017-02-08 by simonqhughes