ARMmbed/mbed-os #3668: STM rtc_isenabled() returns false positive

mfiore02, 2017-01-31:
### Description
- Type: Bug
- Priority: Minor

The mbed RTC API used to allow calls to time() without a preceding set_time() call. The API would detect that the RTC was not yet initialized or running and initialize it before returning the time to ensure that the returned time was valid. This is no longer true for STM targets. If the platform has DEVICE_LOWPOWERTIMER defined, the RTC behaves as expected, but otherwise the RTC is not initialized unless set_time() is explicitly called because rtc_isenabled() unconditionally returns success in this case.

This should be fixed or the change in behavior explicitly documented. I'm unsure as to why the RTC_ISR is only checked if DEVICE_LOWPOWERTIMER is defined. @bcostm, is this not also sufficient to determine if the RTC is running when DEVICE_LOWPOWERTIMER is undefined? I tested this on XDOT_L151CC platform and it appears to work.

I understand that adding a set_time() call resolves this, but I'm concerned about subtle changes in behavior like this breaking applications that were unknowingly (or knowingly) relying on them.

I will submit a PR adding a RTC unit test for this case. I will also look into submitting a PR with a fix for this issue.

---------------------------------------------------------------
## Bug

**Target**
Tested on XDOT_L151CC. Affects any STM target which has DEVICE_RTC defined and DEVICE_LOWPOWERTIMER undefined.

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
0.9.1

**meed-os sha:**
269f58d Merge pull request #3660 from ARMmbed/release-candidate

**DAPLink version:**
0241

**Expected behavior**
When calling time(NULL) without a previous set_time() call, the RTC API should properly detect that RTC is not running and initialize it.

**Actual behavior**
If DEVICE_LOWPOWERTIMER is undefined, rtc_isenabled() simply returns 1, causing time() calls without a preceding set_time() call to return a constant, invalid value.

**Steps to reproduce**
Building and running the following application on a MTS xDot or any other ST platform that defines DEVICE_RTC but not DEVICE_LOWPOWERTIMER should cause a constant invalid time to get printed out the debug port. Uncommenting the set_time() line resolves the issue.

```C++
#include "mbed.h"

Serial pc(USBTX, USBRX);

int main() {
    //set_time(1256729737);

    while (true) {
        printf("time %d\r\n", time(NULL));
        wait_ms(2000);
    }
}
```

jeromecoutant, 2017-01-31:
Hi
I am currently working on this kind of issue.
Regards,

mfiore02, 2017-01-31:
@jeromecoutant can you elaborate? Are you fixing this issue? If so, are you also planning on adding the unit test I mentioned?

mfiore02, 2017-01-31:
It may not be trivial to add a unit test for this since the RTC must be in an uninitialized state for this scenario to be properly tested.

jeromecoutant, 2017-01-31:
I made some "private" unit tests :-)
And I agree that the DEVICE_LOWPOWERTIMER check is not accurate.

My current code is something like this:
return ( ((RTC->ISR & RTC_ISR_INITS) ==  RTC_ISR_INITS) && ((RTC->ISR & RTC_ISR_RSF) ==  RTC_ISR_RSF) );

mfiore02, 2017-01-31:
I've been attaching a custom rtc_isenabled() in my applications with contents very similar to that. It's been working well to detect if the RTC is running even after waking from stop or standby modes on both the F411RE and L151CC devices.

```C++
int dot_is_rtc_enabled(void) {
    return (RTC->ISR & RTC_ISR_INITS ) && (RTC->ISR & RTC_ISR_RSF ); 
}
```

jeromecoutant, 2017-03-01:
Hi
Could we close this issue ?
Thx

closed 2017-03-01 by 0xc0170