ARMmbed/mbed-os #3680: 'cmsis' exporter is broken in the online compiler (NUCLEO_F411RE)

janjongboom, 2017-02-02:
Blinky (mbed OS 5) on the NUCLEO_F411RE, export to 'cmsis' -> 'Export failed'.

Actually I don't even know what the 'cmsis' exporter is :-).

Works in mbed CLI.

0xc0170, 2017-02-02:
cc @theotherjimmy

theotherjimmy, 2017-02-02:
@janjongboom 

> Blinky (mbed OS 5) on the NUCLEO_F411RE, export to 'cmsis' -> 'Export failed'.

I'm hunting it down in sentry

>Actually I don't even know what the 'cmsis' exporter is :-).

The CMSIS team defined a project file format a while back. This exporter is that project file.

> Works in mbed CLI.

I'll look at the diff.

theotherjimmy, 2017-02-02:
> I'm hunting it down in sentry

It's actually caught. The exception is Export failed: join() takes at least 1 argument (0 given)

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

closed 2017-10-27 by ChiefBureaucraticOfficer

janjongboom, 2017-10-30:
@theotherjimmy Has this been resolved?