ARMmbed/mbed-os #3681: Issues with the fixed heap size in IAR

bogdanm, 2017-02-02:
I compiled a project for K64F with three compilers (GCC_ARM, ARMCC and IAR). The binaries compiled with GCC_ARM and ARMCC worked fine,  but the one compiled with IAR failed with an out of memory error. After further investigation, it appears that IAR linker command file declares a maximum size for the heap, which is enforced at runtime:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_IAR/MK64FN1M0xxx12.icf#L85
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_IAR/MK64FN1M0xxx12.icf#L104

This isn't consistent with GCC_ARM and ARMCC, which check for a heap/stack colision first, and only then with a dynamically computed heap limit (https://github.com/ARMmbed/mbed-os/blob/master/platform/retarget.cpp#L596).

So the question is, what can be done to make this consistent in all three toolchains? As a quick fix, it appears that it's possible to declare the heap without a maximum size in IAR:

```
define block HEAP      with alignment = 8     { };
```

The above change fixes my particular issue, but I'm not sure about its full implications.

@marcuschangarm

marcuschangarm, 2017-02-02:
@JanneKiiskila @yogpan01 @TeroJaasko @teetak01 

FYI

bogdanm, 2017-02-02:
Apparently I misread the preprocessor guards in `retarget.cpp`,  `_sbrk` is implemented only for GCC, not ARMCC. Still, ARMCC seems to do the same thing as GCC.

bulislaw, 2017-02-03:
CC: @c1728p9

JanneKiiskila, 2017-02-03:
@kuggenhoffen  - I think this might solve the issue we have had with mbed-client-testapp?

TeroJaasko, 2017-02-03:
This issue seems to be a duplicate of the one which Jaakko reported last month: https://github.com/ARMmbed/mbed-os/issues/3466

c1728p9, 2017-02-04:
Hi @bogdanm I gave that same modification a try. In the map file it creates the heap region has a size of 0. Malloc still returns a valid pointer but it looks as if bounds checking isn't working. If called repeatedly malloc will keep returning a non-null pointer until the program crashes. If a HEAP non-zero region size is specified then malloc eventually returns null as expected.

Fixed size:
```
"P5":                                        0x10000
  HEAP                           0x20002318  0x10000  <Block>
    HEAP                uninit   0x20002318  0x10000  <Block tail>
                               - 0x20012318  0x10000
```

No size:
```
"P5":                                           0x0
  HEAP                           0x1fff2718     0x0  <Block>
```

0xc0170, 2017-02-06:
> This issue seems to be a duplicate of the one which Jaakko reported last month: #3466

Thanks for referencing. as it stated there, this inconsistency is there are IAR does not support dynamic heap.

closed 2017-10-27 by ChiefBureaucraticOfficer