ARMmbed/mbed-os #3682: Online compiler: Clone from GH -> publish to mbed -> always fails

janjongboom, 2017-02-02:
1. Open the online compiler
2. Import a project from GitHub
3. Change some stuff and commit
4. Publish
5. Click 'Fork...' and put under an mbed.org organisation
6. Click 'Publish'
7. An error occurs
8. The project is actually published though, but you need to re-import this project to be able to commit to it, which is very annoying

sg-, 2017-02-07:
Publishing of git repos was suppose to be blocked until further credential integrations was completed. @screamerbg was something deployed?

janjongboom, 2017-02-07:
@sg- This is not about publishing git repos. It's about:

* Import a Git repo
* Fork it to mbed.org

So it goes from git -> hg.

I wouldn't care about losing history, just want it published on mbed.org.

sg-, 2017-02-08:
Pretty sure thats even less of a desired use case 😆

janjongboom, 2017-02-08:
@sg- @screamerbg If we would just have a 'New repository' button on the teams page that would help me already, because then I can do this all locally. But as far as I know the only way to make a new repo is through the compiler...

sg-, 2017-02-08:
This button?

[image]

sg-, 2017-02-08:
or this one?

[image]

janjongboom, 2017-02-09:
@sg- Ah wow. And there I was banging my head against the wall because I was expecting this button on the dashboard next to my repos, not on the Code tab.

[image]

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

janjongboom, 2017-10-30:
@ChiefBureaucraticOfficer This is still an issue.

closed 2019-10-31 by ciarmcom