ARMmbed/mbed-os #3683: cmsis5/rtx2: ST: execution never leaves SystemInit

bulislaw, 2017-02-02:
### Description
- Type: Bug 
- Priority: Blocker
- Branch: https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5

---------------------------------------------------------------
## Bug

**Target**
ST platfoms (tested with NUCLEO F401RE and F411RE)

**Toolchain:**
GCC_ARM ARM

We're trying to finalize CMSIS5/RTX2 (https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5) update in next month or two, but we're having issues with ST platforms. When executing tests (eg `mbed test -n tests-mbedmicro-rtos-mbed-basic -m NUCLEO_F401RE -vv`) I'm getting a timeout. I had a bit of troubles trying to debug the issue, but as far as i can tell the execution never leaves `SystemInit`, it's stuck in `TIM5_IRQHandler` in an infinite loop. While stepping through the code it seems the interrupt was happening while trying to access SysTick counter (`HAL_GetTick`), but I'm not 100% sure.

@0xc0170 Could you ping some ST people, maybe they could have a look into it?
CC: @sg- @c1728p9

0xc0170, 2017-02-02:
cc @bcostm @adustm @LMESTM @jeromecoutant

bcostm, 2017-02-03:
Hi,
This test is ok on the master branch. I don't know if you can rebase your branch on top of master ?

```
mbedgt: test suite report:
+-----------------------+---------------+---------------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                      | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+---------------------------------+--------+--------------------+-------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-basic | OK     | 22.23              | shell       |
+-----------------------+---------------+---------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-----------------------+---------------+---------------------------------+---------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                      | test case                       | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+---------------------------------+---------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-basic | tests-mbedmicro-rtos-mbed-basic | 1      | 0      | OK     | 22.23              |
+-----------------------+---------------+---------------------------------+---------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 OK
mbedgt: completed in 26.92 sec

mbed-os version
65956d1 Merge pull request #3677 from c1728p9/armcc_linker_flags
```

bulislaw, 2017-02-03:
I know it is ok on master and the feature branch is almost up to date with master anyway. But it seems to be broken (for all ST platforms I've tried) with the update to cmsis5/rtx2 and my debugging led me into bowels of the platform. It's the change we'll be landing soon on master and I'd like to have ST boards passing all the tests. Could you guys help us with debugging it please.

bulislaw, 2017-02-13:
ping

bcostm, 2017-02-16:
Hi,

I have installed the latest uVision V5.23 toolchain and generated a project using "mbed export" in order to start debugging.

But I have these 2 errors:
```
assembling irq_cm4f.s...
rtos\rtx2\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\irq_cm4f.s(81): error: A1854E: Unknown opcode 'VSTMDBEQ', maybe wrong target CPU?
rtos\rtx2\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM\irq_cm4f.s(97): error: A1854E: Unknown opcode 'VLDMIAEQ', maybe wrong target CPU?
```

It's strange because when I compile with `mbed test -vv -m NUCLEO_F401RE -t ARM --compile` I don't have any error. Same with GCC_ARM build, no error.

Do you use a uvision project for your debug ?

bulislaw, 2017-02-16:
I don't generally use windows, but yes when I was trying to debug this issue I was using uVision 5.20 (ARMC 5.06). It seems that the flag is correctly set `__FPU_PRESENT` as this platform has FPU. I'm not in the office and don't have access to windows machine right now to recheck that, can uVision  debugger use binary built externally by `mbed compile --profile debug -m NUCLEO_F401RE -t ARM`?

bcostm, 2017-02-16:
I don't know if this is feasible or not ? Anyway I have commented the 2 lines just to see if the rest was ok and I have now 4247 errors during linking...
But seems to be the same error:

```
*** Using Compiler 'V5.06 update 4 (build 422)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'mbed-os'
compiling main_blink_led.cpp...
linking...
.\BUILD\mbed-os.axf: Error: L6366E: analogin.o attributes are not compatible with the provided attributes .
Object analogin.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os.axf: Error: L6367E: analogin.o(.rev16_text) attributes are not compatible with the provided attributes .
Object analogin.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os.axf: Error: L6367E: analogin.o(.revsh_text) attributes are not compatible with the provided attributes .
Object analogin.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
.\BUILD\mbed-os.axf: Error: L6367E: analogin.o(.rrx_text) attributes are not compatible with the provided attributes .
Object analogin.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
```
...

bulislaw, 2017-02-16:
Do you have similar issues on `master`?

bcostm, 2017-02-16:
Yes same error on master.

bulislaw, 2017-02-16:
OK, it explains a lot I did a rebase on monday. You mind creating a bug report for master?

bulislaw, 2017-02-20:
@bcostm I manage to reproduce the issue with current master and only CMSIS changed, I prepared branch for F401 https://github.com/bulislaw/mbed-os/tree/st401_update_cmsis_no_rtx
Tested with tests-mbedmicro-rtos-mbed-threads (which hangs)

bcostm, 2017-02-21:
I have now a uvision project that compiles thanks to the fix Martin suggested.

I have tested your branch `st401_update_cmsis_no_rtx` with different test codes:
- First, with a blinky example + uVision ->OK. Note that using this simple code on the `feature_cmsis5` branch doesn't work for me (hang).
- Then with tests-mbedmicro-rtos-mbed-basic code (remove Greentea code) + uVision -> OK
- Then with the tests-mbedmicro-rtos-mbed-threads code + uVision ->OK
- Launch thread test with `mbed test`
```
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite                        | test case                            | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads             | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with child  | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with murder | 1      | 0      | OK     | 0.08               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with wait   | 1      | 0      | OK     | 0.16               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with yield  | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads               | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with child    | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with murder   | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with wait     | 1      | 0      | OK     | 1.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with yield    | 1      | 0      | OK     | 0.08               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread                | 1      | 0      | OK     | 0.03               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with child     | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with murder    | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with wait      | 1      | 0      | OK     | 0.19               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with yield     | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing thread self terminate        | 1      | 0      | OK     | 0.09               |
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
```
- Then launch all rtos tests:
```
+-------------------+---------------+-------------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite                          | test case                            | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+-------------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-basic     | tests-mbedmicro-rtos-mbed-basic      | 0      | 1      | ERROR  | 69.44              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-isr       | tests-mbedmicro-rtos-mbed-isr        | 1      | 0      | OK     | 16.35              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-mail      | tests-mbedmicro-rtos-mbed-mail       | 1      | 0      | OK     | 12.89              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-malloc    | tests-mbedmicro-rtos-mbed-malloc     | 1      | 0      | OK     | 26.32              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-mutex     | tests-mbedmicro-rtos-mbed-mutex      | 1      | 0      | OK     | 21.47              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-queue     | tests-mbedmicro-rtos-mbed-queue      | 1      | 0      | OK     | 12.94              |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-semaphore | tests-mbedmicro-rtos-mbed-semaphore  | 1      | 0      | OK     | 18.68              |
[... cut ...]
[1487690390.41][HTST][WRN] missing __exit event from DUT
[1487690390.41][HTST][WRN] missing __exit_event_queue event from host test
[1487690390.41][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1487690390.41][HTST][INF] calling blocking teardown()
[1487690390.41][HTST][INF] teardown() finished
[1487690390.41][HTST][INF] {{result;timeout}}
```

Are you sure it hangs for you ? or maybe I misinterpreted what you wrote and the `st401_update_cmsis_no_rtx` branch is ok ?

Do you confirm my results ?

bulislaw, 2017-02-21:
I retested the `st401_update_cmsis_no_rtx` branch for both ARM and GCC (using `tests-mbedmicro-rtos-mbed-threads`)  and it works with ARM and hangs with GCC. But never mind that really, if you have way of debugging the `feature_cmsis5` branch now, using Martins workaround, it's better to work with that as that what we care about anyway.

bcostm, 2017-02-22:
Yes I tried also yesterday to debug the feature_cmsis5 branch with a simple blinky program but I didn't find anything...

bcostm, 2017-02-22:
Humm... I have retested this morning the `feature_cmsis5` branch (`65606d7` Merge branch 'master' into feature_cmsis5) and the rtos-threads test is ok for both ARM and GCC:
```
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite                        | test case                            | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads             | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with child  | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with murder | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with wait   | 1      | 0      | OK     | 0.15               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with yield  | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with child    | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with murder   | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with wait     | 1      | 0      | OK     | 1.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with yield    | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread                | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with child     | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with murder    | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with wait      | 1      | 0      | OK     | 0.16               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with yield     | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing thread self terminate        | 1      | 0      | OK     | 0.06               |
+-------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
+-----------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                        | test case                            | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads             | 1      | 0      | OK     | 0.04               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with child  | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with murder | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with wait   | 1      | 0      | OK     | 0.16               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing parallel threads with yield  | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with child    | 1      | 0      | OK     | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with murder   | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing serial threads with wait     | 1      | 0      | OK     | 1.06               |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with wait      | 1      | 0      | OK     | 0.15               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing single thread with yield     | 1      | 0      | OK     | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedmicro-rtos-mbed-threads | Testing thread self terminate        | 1      | 0      | OK     | 0.06               |
+-----------------------+---------------+-----------------------------------+--------------------------------------+--------+--------+--------+--------------------+
```

bcostm, 2017-02-22:
I have generated a uVision project from the feature_cmsis5 branch. I have these warnings for non-declared functions:
rtos\mbed_boot.c(296): warning:  #223-D: function "__rt_lib_init" declared implicitly
rtos\mbed_boot.c(310): warning:  #223-D: function "__user_setup_stackheap" declared implicitly
rtos\mbed_boot.c(313): warning:  #223-D: function "_platform_post_stackheap_init" declared implicitly
rtos\mbed_rtx_handlers.c(25): warning:  #223-D: function "rtos_idle_loop" declared implicitly

Other thing: When I compile the rtos-basic test main.cpp, I have an error due to the DEFAULT_STACK_SIZE  not declared. In the master branch this is defined in the rtos/rtx/target_cortex_m/cmsis_os.h file. But I didn't find it in the feature_cmsis5 branch ?

0xc0170, 2017-02-22:
@bcostm we will fix those warnings, I was planning to send a patch

bulislaw, 2017-02-22:
> Other thing: When I compile the rtos-basic test main.cpp, I have an error due to the DEFAULT_STACK_SIZE not declared. In the master branch this is defined in the rtos/rtx/target_cortex_m/cmsis_os.h file. But I didn't find it in the feature_cmsis5 branch ?

The DEFAULT_STACK_SIZE was removed, I have fix for the tests, I'll merge it soon.

As for the test results, I'm getting timeouts for all tests for `feature_cmsis5 65606d`, Martin is checking for his setup. Could you have a look at the timeouting tests in the meantime.

bulislaw, 2017-02-27:
@0xc0170 any news?

bulislaw, 2017-02-27:
@bcostm I've got some results, that looks similar to yours. Could you please have a look at the failures. I think the issue with building tests is resolved now.

ARM:
```
+-------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target            | platform_name | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+-------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 9.85               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.94               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 12.06              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 18.09              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 11.89              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.91               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 11.34              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 10.89              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 11.03              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 10.65              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | TIMEOUT | 68.46              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 9.89               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 10.87              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 10.2               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | rtos-rtx2-target_cortex_m-tests-memory-heap_and_stack                        | TIMEOUT | 67.85              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-events-queue                                                           | OK      | 17.46              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-integration-basic                                                      | TIMEOUT | 67.72              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-c_strings                                                 | OK      | 11.81              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-dev_null                                                  | TIMEOUT | 68.0               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-echo                                                      | OK      | 28.7               | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-generic_tests                                             | OK      | 10.13              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-race_test                                                 | OK      | 11.22              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-rtc                                                       | OK      | 20.61              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-stl_features                                              | OK      | 11.69              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-ticker                                                    | OK      | 43.06              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-timeout                                                   | OK      | 20.53              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_drivers-wait_us                                                   | OK      | 20.87              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_functional-callback                                               | OK      | 11.72              | shell       |
| NUCLEO_F401RE-ARM | NUCLEO_F401RE | tests-mbed_functional-callback_big                                           | OK      | 11.48              | shell       |
[... cut ...]
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbedtls-selftest                                                       | OK      | 15.52              | shell       |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-netsocket-ip_parsing                                                   | OK      | 12.37              | shell       |
+-----------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
```

bcostm, 2017-02-27:
OK no more error due to the DEFAULT_STACK_SIZE, I can compile the project but I still have these warnings:
```
rtos\mbed_boot.c(294): warning:  #223-D: function "__rt_lib_init" declared implicitly
rtos\mbed_boot.c(308): warning:  #223-D: function "__user_setup_stackheap" declared implicitly
rtos\mbed_boot.c(311): warning:  #223-D: function "_platform_post_stackheap_init" declared implicitly
rtos\mbed_rtx_handlers.c(25): warning:  #223-D: function "rtos_idle_loop" declared implicitly
```

bcostm, 2017-03-02:
We've found something that seems to fix the issue (thanks @jamike) but we need first to launch a full non-regression tests on all our platforms to see if this doesn't break anything else... I'll keep you updated.

bulislaw, 2017-03-08:
@bcostm any news?

bcostm, 2017-03-09:
Hi,
Can you check with this PR ?

bulislaw, 2017-03-15:
That's fixed now.

closed 2017-03-15 by bulislaw