ARMmbed/mbed-os #3687: Fail to set BP in Keil uVision debugger for mbed CLI generated ELF

ccli8, 2017-02-03:
### Description
- Type: Question
- Priority: Minor

## Question
I would like to debug mbed CLI generated ELF file with toolchain set to *ARM* in Keil uVision debugger. This is useful as some issue cannot re-produce thru exported uVision project. For example, I would like to debug *mbed-os-example-blinky*. With build command line below, I get *mbed-os-example-blinky.elf*.
```
mbed compile -t ARM -m NUMAKER_PFM_NUC472 --profile mbed-os/tools/profiles/debug.json
```

Following the link below to generate a dummy Keil uVision project, e.g. *debug-mbed*. I can flash *mbed-os-example-blinky.elf*, enter debug session, and start source-level debugging in this dummy project.
http://www.keil.com/support/docs/2310.htm
In debug session, I can step-run, but fail to set BP in most source files e.g. main.cpp (with **Run to main()** unchecked) until I step into that source file. In my opinion, it would be reasonable for Keil to debug external ELF file generated by itself even thru mbed CLI as long as debug information is included.

### Environment
**Keil uVision**
5.17
**mbed CLI**
1.0.0
**mbed-os-example-blinky**
mbed-os-5.3.4
**mbed-os**
mbed-os-5.3

0xc0170, 2017-02-03:
@ccli8 where do you think the problem is? I remember having this problem with yotta, that did something similar with help of valinor (=generate elf file, open uvision with source files, and point to elf file).  Thus I suspect this is a problem of a tool rather then anything on our side, but that's just an assumption.

cc @theotherjimmy

ccli8, 2017-02-03:
Transfer the issue to mbed CLI repo. Close this one.
https://github.com/ARMmbed/mbed-cli/issues/433

ccli8, 2017-02-03:
Sorry. I re-open the issue. It should be reported here. If I still follow the link below to debug Keil generated .axf file externally rather than mbed CLI generated ELF file, source-level debugging is OK.
http://www.keil.com/support/docs/2310.htm
So I guess compile option or something else provided by mbed-os could fix the issue.

ccli8, 2017-02-06:
Answering my own question, it seems Keil cannot handle well relative path names starting with `".\"` of source files in line to address mapping. With the workaround below, BP can set successfully.

In mbed-os/tools/toolchains/arm.py,
<pre>
def compile(self, cc, source, object, includes):
# Build compile command
cmd = cc + self.get_compile_options(self.get_symbols(), includes)
    
cmd.extend(self.get_dep_option(object))
<b>
# Remove '.\' from relative path names of source files. For example, '.\mbed-os\platform\retarget.cpp' will truncate to 'mbed-os\platform\retarget.cpp'.
if source.startswith('.\\'):
    source = source[2:]
</b>     
cmd.extend(["-o", object, source])
</pre>

closed 2017-02-06 by ccli8

0xc0170, 2017-02-06:
@ccli8 Thanks for the update! Worth having mainlined this patch?

ccli8, 2017-02-07:
@0xc0170 If you suggest the debug trick officially for mbed development, it would be better to mainline the above patch. Otherwise, unnecessary.