ARMmbed/mbed-os #3688: cmsis5/rtx2: ublox_evk_odin_w2 fails to build with ARMC

bulislaw, 2017-02-03:
### Description
- Type: Bug
- Priority: Blocker 
- Branch: https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5
---------------------------------------------------------------
## Bug

**Target**
ublox_evk_odin_w2

**Toolchain:**
ARM

We're trying to finalize CMSIS5/RTX2 (https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5) update in next month or two, but we're having issue with Odin on ARMC (it works fine for GCC though).

```
Error: L6218E: Undefined symbol rtos::Semaphore::Semaphore(int) (referred from cb_main.o).
Error: L6218E: Undefined symbol osThreadCreate (referred from cb_main.o).
Error: L6218E: Undefined symbol osTimerCreate (referred from cb_hw.o).
Finished: 0 information, 0 warning and 3 error messages.
[ERROR] Error: L6218E: Undefined symbol rtos::Semaphore::Semaphore(int) (referred from cb_main.o).
Error: L6218E: Undefined symbol osThreadCreate (referred from cb_main.o).
Error: L6218E: Undefined symbol osTimerCreate (referred from cb_hw.o).
Finished: 0 information, 0 warning and 3 error messages.
```

@andreaslarssonublox Could you guys, update your code for RTX2 and regenerate binaries and submit PR against the feature_cmsis5 please.

bulislaw, 2017-02-21:
@andreaslarssonublox ping

bulislaw, 2017-03-08:
@andreaslarssonublox any chance you could have a look at that? Also @0xc0170 got IAR support, there are still some issues with it, but generally it should be possible to figure out if it will work or not.

andreaslarssonublox, 2017-03-08:
Yes, @ramtin-r is looking at it right now.

bulislaw, 2017-03-15:
@ramtin-r any updates?

andreaslarssonublox, 2017-03-15:
@bulislaw We got it running for GCC_ARM and ARM but with IAR we could not compile. Is it supposed to work for IAR now?

Got the following errors with IAR for EventQueue.cpp:
       [Error] stm32f4xx_hal_def.h@132,12: [Pe020]: identifier "__LDREXW" is undefined
       [Error] stm32f4xx_hal_def.h@134,25: [Pe020]: identifier "__STREXW" is undefined
       [Error] stm32f4xx_hal_def.h@142,12: [Pe020]: identifier "__LDREXW" is undefined
       [Error] stm32f4xx_hal_def.h@144,0: [Pe020]: identifier "__STREXW" is undefined

bulislaw, 2017-03-15:
Ah, we got the same issue somewhere else. @0xc0170 is working on that.

0xc0170, 2017-03-16:
> @bulislaw We got it running for GCC_ARM and ARM but with IAR we could not compile. Is it supposed to work for IAR now?

> Got the following errors with IAR for EventQueue.cpp:
[Error] stm32f4xx_hal_def.h@132,12: [Pe020]: identifier "__LDREXW" is undefined
[Error] stm32f4xx_hal_def.h@134,25: [Pe020]: identifier "__STREXW" is undefined
[Error] stm32f4xx_hal_def.h@142,12: [Pe020]: identifier "__LDREXW" is undefined
[Error] stm32f4xx_hal_def.h@144,0: [Pe020]: identifier "__STREXW" is undefined

@andreaslarssonublox To fix the above identifier LDREXW undefined, please have a look at https://github.com/ARMmbed/mbed-os/pull/3946

bulislaw, 2017-04-20:
@andreaslarssonublox @ramtin-r ping

we've switched the WiFi off for this platform to avoid build errors, but it's not a good solution.

andreaslarssonublox, 2017-04-20:
Oki, yes we would prefer to generate new binaries once all tests pass for all toolchains. We tested it yesterday and then the WiFi tests worked for all toolchains(with some fixes). After upgrading to the latest mbed-os in the branch we don't pass the tests anymore. We are currently investigating why it has stopped working.

0xc0170, 2017-04-20:
Thanks @andreaslarssonublox , keep us updated.

> After upgrading to the latest mbed-os in the branch we don't pass the tests anymore

However, what does this mean? What sha did you use?

andreaslarssonublox, 2017-04-20:
This does not work:
fcbd6645b07997da338eb4e273a1c5b778c84876

This works fine:
604d431f6a25fafc3917835003e732a058b00bc4

0xc0170, 2017-04-21:
@andreaslarssonublox One cmsis inclusion breaks the app for you? Have you found out what broke?

bulislaw, 2017-05-03:
Guys it's actually a blocker for us now, i have tried to work around it here https://github.com/ARMmbed/mbed-os/pull/4264
but i'm getting `Error: L6218E: Undefined symbol rtos::Semaphore::Semaphore(int) (referred from cb_main.o).`

0xc0170, 2017-05-03:
> Guys it's actually a blocker for us now, i have tried to work around it here #4264
but i'm getting Error: L6218E: Undefined symbol rtos::Semaphore::Semaphore(int) (referred from cb_main.o).

The error looks legit ? The ``feature_cmsis5`` branch extended the argument - adding ``max_count``. Shouldn't we let the old one in there and add a new overload with max_count - that way it should be backward compatible?

C++ layer should stay backward compatible

bulislaw, 2017-05-03:
It is compatible, the second parameter has a default value. But it may be a workaround to add second constructor.

andreaslarssonublox, 2017-05-05:
We tested today with the latest code in the CMSIS 5 branch with our driver and it seems to work OK for all toolchains now. We aim to build a new binary for the master branch which should also work for this branch if possible. There should not be any dependencies towards lower layer stuff. This binary will be released within the next two coming weeks.

bulislaw, 2017-05-05:
Thanks, yeah it works as we added RTX1 compatibility layer to simplify the update process.

closed 2017-05-05 by bulislaw