ARMmbed/mbed-os #3690: cmsis5/rtx2 nrf51: RTX SysTick API needs updating

bulislaw, 2017-02-03:
### Description
- Type: Bug 
- Priority: Blocker 
- Branch: https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5

---------------------------------------------------------------
## Bug

**Target**
nrf51_dk (but most likely all Nordic platforms requiring non standard systick)
 
We're trying to finalize CMSIS5/RTX2 (https://github.com/ARMmbed/mbed-os/tree/feature_cmsis5) update in next month, but we're having issues with Nordic platform. When executing tests (eg `mbed test -n tests-mbedmicro-rtos-mbed-basic -m nrf51_dk -vv`) I'm getting a timeout. That's most likely due to changes in RTX API (https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/rtx2/TARGET_CORTEX_M/rtx_system.c#L212) Nordic HAL SysTick override needs to be updated (and validated) for CMSIS5/RTX2.

@pan- Can we ask Nordic guys to help us with the update? I'd rather have someone that knows the platform doing that as it's not enough just to rename the functions.

pan-, 2017-02-03:
@bulislaw Is it possible to have documentation ? I don't see which function have to be called by the tick generator.

bulislaw, 2017-02-03:
There's no documentation in CMSIS5 repo I could find. But as far as i understand 'no function' is called, you just need to rise irq which number you return from `osRtxSysTimerSetup` call.

bulislaw, 2017-02-03:
Oh no, there's 'something': https://github.com/ARM-software/CMSIS_5/blob/155de2a4f69758c885b7df1e3e02f0c57ff7cc55/CMSIS/DoxyGen/RTOS2/src/cmsis_os2.txt#L1569

pan-, 2017-02-03:
Looking at the assembly code, the expectations are a bit different as well. Previously it was just suppose to call the tick handler, now it should call also call `SVC_Context` (see [here](https://github.com/ARM-software/CMSIS_5/blob/155de2a4f69758c885b7df1e3e02f0c57ff7cc55/CMSIS/RTOS2/RTX/Source/GCC/irq_cm0.S#L152-L158)). 

To have help from Nordic engineers it would be better to have a porting guide like for [RTX v1](http://www.keil.com/pack/doc/CMSIS/RTOS/html/timerTick.html). Could we ask this to the RTX team ?

0xc0170, 2017-02-06:
> To have help from Nordic engineers it would be better to have a porting guide like for RTX v1. Could we ask this to the RTX team ?

cc @ReinhardKeil @RobertRostohar - please have a look at this issue

pan-, 2017-02-06:
@bulislaw From my understanding: 

```c++
/**
 * Setup timer (including IRQ priority) and return IRQ number.
 * Was previously int os_tick_init (void)
 */
int32_t  osRtxSysTimerSetup   (void);  

/**
 * Acknowledge IRQ
 * Was: void os_tick_irqack(void)
 */
void     osRtxSysTimerAckIRQ  (void);  // 

/**
 * Current Tick count (32-bit)
 * was uint32_t os_tick_val(void)
 */
uint32_t osRtxSysTimerGetCount(void);  

// These three functions are new functions specific to RTX v2 and have to be implemented
uint32_t osRtxSysTimerGetFreq (void);  // Timer Tick frequency
void     osRtxSysTimerEnable  (void);  // Start timer
void     osRtxSysTimerDisable (void);  // Stop timer

// The function os_tick_ovf is deprecated ...
```
The function `SysTick_Handler ` should be called in the shared RTC IRQ handler instead of `OS_Tick_Handler`.

Do we have the same understanding ?

bulislaw, 2017-02-06:
@0xc0170 we have an email thread about that.
@pan- yes that's exactly what I'm thinking

bulislaw, 2017-02-06:
From the email exchange:

```
You are right, we need to improve RTX5 documentation regarding alternate SysTick timer.
 
Until then the following hopefully helps.
 
The following functions need to be tailored to the alternate tick timer:
 
int32_t  osRtxSysTimerSetup   (void);  // Setup timer (including IRQ priority) and return IRQ number
void     osRtxSysTimerEnable  (void);  // Start timer
void     osRtxSysTimerDisable (void);  // Stop timer
void     osRtxSysTimerAckIRQ  (void);  // Acknowledge IRQ
uint32_t osRtxSysTimerGetCount(void);  // Current Tick count (32-bit)
uint32_t osRtxSysTimerGetFreq (void);  // Timer Tick frequency
 
Names should already explain what the functions should do (additional comments listed).
 
Currently only the IRQ priority needs to be configured which should match PendSV. This can be achieved by ExtTick_SetupIRQ() function (core_cm.h). In the future this will be automatically called by the system. Enabling/disabling IRQ is already done by the system.
 
There is no need to provide an IRQ handler but rather the existing SysTick_Handler should be mapped to the tick IRQ.
```

```
 
·        With RTX v1, the timer was setup and started  at the same time by the function os_tick_init.
Could you confirm that these two behaviours have been split in two function and that it is not required to start the timer in osRtxSysTimerSetup ?
 
·        Is it a defined behaviour to call osRtxSysTimerGetCount or osRtxSysTimerGetFreq  while the timer is disabled ?
 
·        In RTX v1, the function reporting the tick value (os_tick_val) had to return values in descending order, is it still the case with osRtxSysTimerGetFreq ?
```

```
·         Yes,  it is not required to start the timer in osRtxSysTimerSetup.
·         Yes, osRtxSysTimerGetCount or osRtxSysTimerGetFreq  can be called also when timer is disabled and should provide valid values.
·         No,  osRtxSysTimerGetCount needs to provide a free running incremental value over the entire 32-bit range
 ```

pan-, 2017-02-07:
@nvlsianpu @anangl Do you think you can look at this item ? I can help you in the process.

nvlsianpu, 2017-02-13:
On the first sight it's look like nRF51 targets are touched. nRF52 has and using SysTick for RTX.

nvlsianpu, 2017-02-14:
I made PR  https://github.com/ARMmbed/mbed-os/pull/3769 (preview as far). I can't to build NRF51_DK target due to errors form base branch. Could you resolve these errors?

pan-, 2017-02-14:
@nvlsianpu Could you detail the issue you encounter when you build for the NRF51_DK (compiler, commit id, etc) ?

nvlsianpu, 2017-02-14:
GCC_RM, SHA-1: 942bad4396dabd78a306b60d9e5903dae5be0d40
Errors which I encountered:
`
Compile [ 52.1%]: rtx_evflags.c
[DEBUG] Compile: C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q1\bin\arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-
initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-po
inter -Os -mcpu=cortex-m0 -mthumb -DDEVICE_ERROR_PATTERN=1 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDE
VICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -D
TARGET_CORTEX_M -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DTARGET_MCU_NRF51_UNIFIED -DDEVICE_INTERRUPT
IN=1 -D__CORTEX_M0 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -
DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DMBED_BUILD_TIMESTAMP=1487069784.9 -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_AR
M -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DS130 -DTARGET_LIKE_CO
RTEX_M0 -DTARGET_MCU_NRF51_32K @.\BUILD\NRF51_DK\GCC_ARM\.includes_93c0174975aa202636fee1d029efc779.txt -include .\BUILD\NRF51_DK\GCC_ARM\mbed_config.h -MD -MF .\BUI
LD\NRF51_DK\GCC_ARM\mbed-os\rtos\rtx2\TARGET_CORTEX_M\rtx_evflags.d -o .\BUILD\NRF51_DK\GCC_ARM\mbed-os\rtos\rtx2\TARGET_CORTEX_M\rtx_evflags.o .\mbed-os\rtos\rtx2\T
ARGET_CORTEX_M\rtx_evflags.c
[DEBUG] Return: 1
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s: Assembler messages:
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:677: Error: cannot honor width suffix -- `mov r1,#0'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:679: Error: cannot honor width suffix -- `mov r0,#0'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:727: Error: cannot honor width suffix -- `mov r1,#4'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:729: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:731: Error: cannot honor width suffix -- `mov r5,#4'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:737: Error: cannot honor width suffix -- `mov r1,#3'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:739: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:741: Error: cannot honor width suffix -- `mov r5,#3'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:743: Error: cannot honor width suffix -- `neg r5,r5'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:754: Error: cannot honor width suffix -- `orr r5,r2'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:904: Error: cannot honor width suffix -- `mov r1,#4'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:906: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:908: Error: cannot honor width suffix -- `mov r0,#4'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:915: Error: cannot honor width suffix -- `mov r1,#3'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:916: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:919: Error: cannot honor width suffix -- `mov r0,#3'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:921: Error: cannot honor width suffix -- `neg r0,r0'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:926: Error: lo register required -- `add r0,r0,#12'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:982: Error: cannot honor width suffix -- `mov r1,#6'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:984: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:986: Error: cannot honor width suffix -- `mov r0,#6'
[DEBUG] Output: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:987: Error: cannot honor width suffix -- `neg r0,r0'
Traceback (most recent call last):
  File "C:\mbed\mbed-os-example-ble\BLE_LED\mbed-os\tools\make.py", line 293, in <module>
    toolchain))
  File "C:\mbed\mbed-os-example-ble\BLE_LED\mbed-os\tools\build_api.py", line 452, in build_project
    resources.inc_dirs)
  File "C:\mbed\mbed-os-example-ble\BLE_LED\mbed-os\tools\toolchains\__init__.py", line 826, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\mbed\mbed-os-example-ble\BLE_LED\mbed-os\tools\toolchains\__init__.py", line 889, in compile_queue
    raise ToolException(err)
ToolException: C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s: Assembler messages:
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:677: Error: cannot honor width suffix -- `mov r1,#0'
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:679: Error: cannot honor width suffix -- `mov r0,#0'
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:727: Error: cannot honor width suffix -- `mov r1,#4'
[... cut ...]
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:984: Error: cannot honor width suffix -- `neg r1,r1'
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:986: Error: cannot honor width suffix -- `mov r0,#6'
C:\Users\ANPU~1.NVL\AppData\Local\Temp\cce4KX2O.s:987: Error: cannot honor width suffix -- `neg r0,r0'

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\mbed\mbed-os-example-ble\BLE_LED\mbed-os\tools\make.py -t GCC_ARM -m NRF51_DK --source . --build .\BUILD\NRF51_DK\GCC_ARM -c -v"
in "C:\mbed\mbed-os-example-ble\BLE_LED"
`

bulislaw, 2017-02-22:
@nvlsianpu with current `feature_cmsis5` branch for NRF52_DK I'm getting:
```
+------------------+---------------+-------------------------------------+---------+--------------------+-------------+
| target           | platform_name | test suite                          | result  | elapsed_time (sec) | copy_method |
+------------------+---------------+-------------------------------------+---------+--------------------+-------------+
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-basic     | TIMEOUT | 68.38              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-isr       | OK      | 21.23              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-malloc    | OK      | 29.84              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-mutex     | TIMEOUT | 68.41              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-queue     | OK      | 17.72              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-semaphore | OK      | 15.58              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-signals   | OK      | 21.69              | shell       |
| NRF52_DK-GCC_ARM | NRF52_DK      | tests-mbedmicro-rtos-mbed-threads   | OK      | 22.31              | shell       |
+------------------+---------------+-------------------------------------+---------+--------------------+-------------+
```
could you have a look at the failing tests please

nvlsianpu, 2017-02-22:
ack

nvlsianpu, 2017-02-24:
@bulislaw 
https://github.com/ARMmbed/mbed-os/pull/3839
I used GCC 5.4
All test passed, for ARM compiler as well. IAR still is unable to compile tests - but as I guess this is expected for now.

bulislaw, 2017-02-27:
I'm getting all green for NRF52, but back to issues on for NRF51:
```
+------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| target           | platform_name | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
+------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 13.02              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 12.86              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 14.82              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 20.8               | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 14.8               | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 12.9               | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 13.68              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 13.35              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 14.14              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 13.56              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 14.52              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 16.34              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 12.63              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 13.12              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | rtos-rtx2-target_cortex_m-tests-memory-heap_and_stack                        | OK      | 12.34              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-events-queue                                                           | OK      | 19.72              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-integration-basic                                                      | OK      | 11.95              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-c_strings                                                 | OK      | 14.79              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-dev_null                                                  | OK      | 14.54              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-echo                                                      | OK      | 26.96              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-generic_tests                                             | OK      | 13.33              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-lp_timeout                                                | OK      | 22.65              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-race_test                                                 | OK      | 14.39              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-rtc                                                       | OK      | 23.29              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-stl_features                                              | TIMEOUT | 19.5               | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-ticker                                                    | OK      | 46.06              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-timeout                                                   | OK      | 23.57              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_drivers-wait_us                                                   | OK      | 23.47              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_functional-callback                                               | OK      | 14.51              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbed_functional-callback_big                                           | OK      | 14.29              | shell       |
[... cut ...]
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-mbedtls-selftest                                                       | OK      | 53.56              | shell       |
| NRF51_DK-GCC_ARM | NRF51_DK      | tests-netsocket-ip_parsing                                                   | OK      | 15.08              | shell       |
+------------------+---------------+------------------------------------------------------------------------------+---------+--------------------+-------------+
``` for gcc 5.4, I'll run ARM tomorrow.

bulislaw, 2017-03-02:
All test are passing on feature_cmsis5 branch. Closing.

closed 2017-03-02 by bulislaw