ARMmbed/mbed-os #3694: GR PEACH strange behavior when using InterruptIn

sarahmarshy, 2017-02-03:
```C++
#include "mbed.h"

DigitalOut led1(LED1);
InterruptIn irq_data(P2_14); 

// main() runs in its own thread in the OS
// (note the calls to wait below for delays)
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}

```
This will prevent the LED from blinking. This pin is listed as an acceptable GPIO IRQ pin [here]( https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_RENESAS/TARGET_RZ_A1H/gpio_irq_api.c#L65). It does not print the warning "pinmap not found for peripheral" that would indicate that the pin is not supported for this operation. 

Should it be removed from this list? 

Inspired by this question: https://developer.mbed.org/questions/76869/GR_Peach-libraries-broken/#answer12005

0xc0170, 2017-02-06:
cc @TomoYamanaka

TomoYamanaka, 2017-02-08:
@sarahmarshy 

I will check this status, so please wait a moment.

TomoYamanaka, 2017-02-20:
@sarahmarshy 

Sorry for this late reply.
I identified the cause for this issue. It seems not enough the contents that I modified in https://github.com/ARMmbed/mbed-os/pull/3289.
I will PR against this issue soon, contact you after I PR.

TomoYamanaka, 2017-02-20:
@sarahmarshy 

PR about this issue is here.
https://github.com/ARMmbed/mbed-os/pull/3803

closed 2017-10-27 by ChiefBureaucraticOfficer