ARMmbed/mbed-os #3696: Inflated hex file size when compiling with mbed cli compared to online

shayo, 2017-02-04:
I'm trying transition from mbed-classic to mbed OS5 for a large project I have. My target is NRF51_DK_BOOT. I'm using GCC_ARM.
The mbed-os version I'm testing was this:
https://github.com/ARMmbed/mbed-os/#65956d108e822d50acb617d0f4b0cafdb2a91d5f
and also
https://github.com/ARMmbed/mbed-os/tree/feature-nrf5_dfu_s13x_v2

I am experiencing the following error:

AddressOverlapError: Data overlapped at address 0x3C000

I tried "--profile small" but I still get this error.

The NRF51_DK target compiles, but results in a very large hex (~689K).
Why is it so big? When I compile the same project with the online compiler I get a hex that is ~546KB for NRF51_DK and ~555KB for NRF51_DK_BOOT.

theotherjimmy, 2017-02-08:
@shayo The online compiler uses ARMC5, so comparing with GCC_ARM is an apples to oranges comparison. Could you try compiling with ARMC5 offline to compare sizes?

shayo, 2017-02-09:
Is ARMC5 part of uVision? If it is only a paid product, then I don't have access to it.
This might explain it. I wasn't aware that the backend doesn't use GCC_ARM.

theotherjimmy, 2017-02-10:
>Is ARMC5 part of uVision?

Yes and DS-5

>This might explain it.

I think so too.

Could you close this issue if you think it's resolved?

closed 2017-02-10 by shayo