ARMmbed/mbed-os #3714: Online compiler inflated binary sizes

neilt6, 2017-02-07:
Something changed in the online compiler within the last few weeks that's causing inflated binary sizes for projects built using mbed-dev. When I tried to resume work on one of my projects that's using a local copy of mbed-dev rev. 153, I was surprised to find the project no longer linked. I tried an experiment using the blinky template for an mbed LPC11U24, and the mbed-dev version was a whopping 5.9kB larger than the mbed binary version and failed to run. I checked the build macros to see if the toolchain had changed, but they still specify ARM micro.

0xc0170, 2017-02-07:
cc @theotherjimmy

neilt6, 2017-02-22:
#3813 and #3815 appear to be related.

neilt6, 2017-03-15:
Hey guys, not trying to be rude but is this going to be fixed any time soon? I'm losing lots of development time, and at this point I'm debating investing in an offline toolchain.

theotherjimmy, 2017-03-15:
I have fixed the problem. The fix is in a PR (not here) and pending review and deploy. It might be best for you to invest in an offline toochain in general. You should be able to use GCC ARM Embedded through mbed-cli without issue, but that may not support your targets, if they only use uARM. I think it would be worth setting up the GCC target so that you don't have to worry about online compiler down time internet connection issues when doing development for projects that are supported by GCC.

neilt6, 2017-03-15:
Thanks @theotherjimmy. I've been clinging to the online compiler so far since I've not been able to match the efficiency of the online μARM toolchain. The last time I looked into this, the version of microlib included with Keil didn't even support C++. The online compiler has been pretty reliable for me before this issue.

theotherjimmy, 2017-03-15:
further update @neilt6: The PR has been merged. I'll let you know when it's live.

neilt6, 2017-03-28:
@theotherjimmy: Any update on when the patched toolchain is going to go live?

theotherjimmy, 2017-03-28:
So, my original patch was reverted as it caused some problems, and I'm working on a new one. I'll let you know when that goes live.

theotherjimmy, 2017-04-05:
The new patch is live.

theotherjimmy, 2017-04-05:
Please re-open if it's still a problem

closed 2017-04-05 by theotherjimmy

neilt6, 2017-04-05:
@theotherjimmy Confirmed, projects are compiling again. Thanks!

theotherjimmy, 2017-04-05:
Sorry that took so long. Please feel free to open new issues related to the online compiler on this repo.