ARMmbed/mbed-os #3715: DISCO_F429ZI : code is blocked with ARM and IAR

adustm, 2017-02-07:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: #3397
- Priority: Blocker

---------------------------------------------------------------
## Bug
DISCO_F429ZI does not work any more in ARM and IAR since integration of PR#3397
GCC_ARM is ok.
The startup file has been modified for uvisor integration. HAL_Init and SystemInit have been split into 2 functions. The 2 new functions (HAL_InitPre and SystemInit_Pre) are not called in ARM and IAR

I am creating the PR

**Expected behavior**
mbed tests OK
**Actual behavior**
every tests in timeout
**Steps to reproduce**

bulislaw, 2017-02-08:
@AlessandroA

adustm, 2017-05-05:
Hello, DISCO-F429ZI is tested ok for the 3 toolchains on release mbed-os-5.4.3
@0xc0170, can you close this issue please ?

closed 2017-05-08 by 0xc0170