ARMmbed/mbed-os #3725: cstddef: No such file or directory

Darker, 2017-02-08:
### Description
- Type: Question

---------------------------------------------------------------
## `mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory`

I cloned this repository and tried to compile this program:

```
/**
 * Nucleo program that does not do anything. Test of basic compilation. 
**/

#include <mbed.h>
void main()
{

}
```

With this command:

`arm-none-eabi-gcc main.c --specs=nosys.specs -I../../software/mbed-os/`

Where `../../software/mbed-os/` is path to the root of this repository. The error I get:
```
In file included from ../../software/mbed-os/mbed.h:51:0,
                 from main.c:5:
../../software/mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory
compilation terminated.
```
The version of my compiler:
```
>arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```
I'm running this all on Windws 7 x64. I found some vague mentions of something called `private_settings.py` in some forum threads, but nobody really bothered to explain what is that supposed to do.

theotherjimmy, 2017-02-08:
@Darker We have a hal that includes code and header files for many targets, and many features in the code that can be turned on and off. Further, we have a few more arguments to gcc than just `--specs=nosys.specs`. As such we have a bunch of tools that run the compiler for us. Please use the pip package mbed-cli to compile.

Further, we use GCC version 4.9.3.

Finally, it looks like you are missing system header files. But I'm not sure this is even the real problem here. Could you try changing your GCC version and using mbed-cli first?

toyowata, 2017-02-09:
@Darker The ```cstddef``` is C++ header file.  Can you rename your source file from ```main.c``` to ```main.cpp```?

ChiefBureaucraticOfficer, 2017-10-27:
GitHib issue review: Closed due to inactivity.  Please re-file if critical issues found.

bkht, 2018-01-11:
Hi, I do have this problem:
    $ mbed new mbed-os-program
    $ cd mbed-os-program
    $ mbed ls -a
    $ mbed compile -t GCC_ARM -m NUCLEO_F746ZG -v
    [Fatal Error] platform.h@22,19: cstddef: No such file or directory

    $ sudo mbed config --list
    [mbed] Global config:
    GCC_ARM_PATH=/usr/bin
    ARM_PATH=/home/jack/apps/gcc-arm-none-eabi-7-2017-q4-major

    [mbed] Local config (/home/jack/apps/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include):
    Couldn't find valid mbed program in /home/jack/apps/gcc-arm-none-eabi-7-2017-q4-major/arm-none-eabi/include

What can be the issue?

0xc0170, 2018-01-12:
> GCC_ARM_PATH=/usr/bin

what gcc is there ? the ARM path points to gcc (use GCC 6 as that one is supported). T hus I would fix GCC_ARM_PATH to point to GCC ARM 6 version

bkht, 2018-01-12:
Hi, I now have
    $ arm-none-eabi-gcc --version
    arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
So, I need to install arm-none-eabi-gcc v6 better?

pan-, 2018-01-12:
@bkht  what type of file triggers the error ? C or C++ ?

0xc0170, 2018-01-12:
> On
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
Only version 7 is listed...
Where can I get v6?

scroll down more, 6 is there

bkht, 2018-01-12:
Many thanks!!!

bkht, 2018-01-12:
Hi, I just installed gcc 6.3.1

    $ arm-none-eabi-gcc --version
    arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

    $ sudo mbed config -G ARM_PATH "/home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux"
    $ sudo cp /home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux/bin/arm-none-eabi-gcc /usr/bin/.
    $ sudo mbed config -G GCC_ARM_PATH "/usr/bin"

    $ arm-none-eabi-gcc --version
    6.3.1

    $ sudo mbed config --list
    [mbed] Global config:
    GCC_ARM_PATH=/usr/bin
    ARM_PATH=/home/jack/apps/gcc-arm-none-eabi-6_2-2016q4

    [mbed] Local config (/home/jack/mbed/mbed-os-program):
    TOOLCHAIN=GCC_ARM
    TARGET=NUCLEO_F746ZG

But the result is the same:

    $ mbed compile -v
    [Fatal Error] platform.h@22,19: cstddef: No such file or directory

What else can I verify to correct the problem?

0xc0170, 2018-01-12:
> @bkht what type of file triggers the error ? C or C++ ?

If you can answer this question, where is the error coming from (not only what header file).

bkht, 2018-01-12:
Hi, this is the complete output, hope that helps:
    $ mbed compile -v
    [mbed] Working path "/home/jack/mbed/mbed-os-program" (program)

    [mbed] Exec "/usr/bin/python -u /home/jack/mbed/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F746ZG --source . --build ./BUILD/NUCLEO_F746ZG/GCC_ARM -v" in /home/jack/mbed/mbed-os-program
    Building project mbed-os-program (NUCLEO_F746ZG, GCC_ARM)
    Scan: .
    Scan: env
    Scan: mbed
    Scan: FEATURE_LWIP
    [DEBUG] Macros: -DTARGET_STM32F746xG -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DTARGET_STM32F7 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_ANALOGOUT=1 -DTARGET_STM32F746ZG -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DARM_MATH_CM7 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M7 -DDEVICE_RTC=1 -DDEVICE_TRNG=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M7 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -D__CORTEX_M7 -DTARGET_STM32F746 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DTARGET_NUCLEO_F746ZG -DTARGET_LIKE_MBED -DTARGET_RTOS_M4_M7 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1515771245.08 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1
    Compile [  0.5%]: AnalogIn.cpp
    [DEBUG] Compile: /usr/bin/arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -DTARGET_STM32F746xG -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DTARGET_STM32F7 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_ANALOGOUT=1 -DTARGET_STM32F746ZG -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DARM_MATH_CM7 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M7 -DDEVICE_RTC=1 -DDEVICE_TRNG=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M7 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -D__CORTEX_M7 -DTARGET_STM32F746 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DTARGET_NUCLEO_F746ZG -DTARGET_LIKE_MBED -DTARGET_RTOS_M4_M7 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1515771245.08 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 @./BUILD/NUCLEO_F746ZG/GCC_ARM/.includes_5fa684d087529e8cf7d6ff863af0a169.txt -include ./BUILD/NUCLEO_F746ZG/GCC_ARM/mbed_config.h -MD -MF BUILD/NUCLEO_F746ZG/GCC_ARM/mbed-os/drivers/AnalogIn.d -o BUILD/NUCLEO_F746ZG/GCC_ARM/mbed-os/drivers/AnalogIn.o ./mbed-os/drivers/AnalogIn.cpp
    [Fatal Error] platform.h@22,19: cstddef: No such file or directory
    [DEBUG] Return: 1
    [DEBUG] Output: In file included from ./mbed-os/drivers/AnalogIn.h:19:0,
    [DEBUG] Output:                  from ./mbed-os/drivers/AnalogIn.cpp:17:
    [DEBUG] Output: ./mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory
    [DEBUG] Output:  #include <cstddef>
    [DEBUG] Output:                    ^
    [DEBUG] Output: compilation terminated.
    Traceback (most recent call last):
    File "/home/jack/mbed/mbed-os-program/mbed-os/tools/make.py", line 296, in <module>
        stats_depth=options.stats_depth)
    File "/home/jack/mbed/mbed-os-program/mbed-os/tools/build_api.py", line 533, in build_project
        objects = toolchain.compile_sources(resources, resources.inc_dirs)
    File "/home/jack/mbed/mbed-os-program/mbed-os/tools/toolchains/__init__.py", line 928, in compile_sources
        return self.compile_queue(queue, objects)
    File "/home/jack/mbed/mbed-os-program/mbed-os/tools/toolchains/__init__.py", line 991, in compile_queue
        raise ToolException(err)
    ToolException: In file included from ./mbed-os/drivers/AnalogIn.h:19:0,
                    from ./mbed-os/drivers/AnalogIn.cpp:17:
    ./mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory
    #include <cstddef>
                    ^
    compilation terminated.

    [mbed] ERROR: "/usr/bin/python" returned error code 1.
    [mbed] ERROR: Command "/usr/bin/python -u /home/jack/mbed/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F746ZG --source . --build ./BUILD/NUCLEO_F746ZG/GCC_ARM -v" in "/home/jack/mbed/mbed-os-program"
    ---

theotherjimmy, 2018-01-12:
@bkht You may need `arm-none-eabi-newlib` or similar to get the c std lib.

bkht, 2018-01-13:
How do I install 'arm-none-eabi-newlib'? I searched, but no luck.

pan-, 2018-01-15:
@bkht The following lines are suspicious: 

```
$ sudo cp /home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux/bin/arm-none-eabi-gcc /usr/bin/.
$ sudo mbed config -G GCC_ARM_PATH "/usr/bin"
```

Why do you copy manually `arm-none-eabi-gcc` into `/usr/bin/` ? A compiler is not just a binary, it come with a linker, headers files, object files, library files, ...

If you have extracted GCC in `/home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux/` then you can set `GCC_ARM_PATH` to `/home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux/bin`: 

```
sudo mbed config -G GCC_ARM_PATH "/home/jack/apps/gcc-arm-none-eabi-6-2017-q2-update-linux/bin"
```

pan-, 2018-01-15:
@bkht the `main` function is not defined by mbed OS; it shall be defined by your application. Add the following code in a file named `main.cpp` in the root or your project: 

```c++
#include "mbed.h"

DigitalOut led1(LED1);

// main() runs in its own thread in the OS
int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
}
```

This application should blink your board LED every 500ms.

bkht, 2018-01-16:
Thank you very much!, Now it all works!
I didn't even notice main.ccp was not in there, as I was trying to test my toolchain installation. :-)
Many thanks!

closed 2018-01-16 by pan-