ARMmbed/mbed-os #3728: Exporting from online compiler failing for several NXP platforms (mbed os v5.3.2)

maclobdell, 2017-02-08:
### Description
- Type: Bug
- Related issue: n/a
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
This seems to be a problem for all targets that are supported under: 
targets\TARGET_Freescale\TARGET_MCUXpresso_MCUS\
(K64F, K22, KL43Z, K66, HEXIWEAR, etc)

**Toolchain:**
Exporting to IAR or Eclipse result in error in online compiler.  See below.

Exporting to KDS results in inability to compile because files are missing under TARGET_Freescale.

**Toolchain version:**
n/a

**mbed-cli version:** 
n/a

**meed-os sha:**
mbed os version 5.3.4 (and 5.3.3 and 5.3.2 I suspect)

**DAPLink version:**
n/a

**Expected behavior**
Exporter should generate project files that are usable.

**Actual behavior**

Selecting IAR or Eclipse export options results in online compiler error message "An error occurred during export.  Please contact support."

Selecting KDS or uVision generates a download, but opening it is not usable in KDS because of missing files.  Did not test uVision, but suspect the files will be missing there too.

**Steps to reproduce**
In the online compiler, import mbed-os-example-blinky.  Make sure it is the latest version (v5.3.2 or later).  Select an affected platform.  Right click on project name, choose export, select an affected IDE.

We suspect that the online compiler is not handing the directory changes that were introduced in v5.3.2 which moved NXP platform code from KSDK2 to MCUxpresso folder.  The online compiler exporters should be able to find the files based on targets.json, but it is not.  

cc 
@theotherjimmy 
@mmahadevan108  (this issue is in the online compiler, cc'ing you for information only)

theotherjimmy, 2017-02-08:
Thanks for recording this. I'm looking into it on partnerbeta.

theotherjimmy, 2017-04-26:
Looks like I tracked this down a while back and fixed it. Closing. Please re-open if it's not as fixed as I though.

closed 2017-04-26 by theotherjimmy