ARMmbed/mbed-os #3729: exporter eclipse_gcc_arm: adding size statistics to output

JojoS62, 2017-02-08:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
**Toolchain:**
GCC_ARM

**Toolchain version:**
exporter eclipse_gcc_arm

----------------------------------------------------------------
## Enhancement

The eclipse_gcc_arm exporter uses a makefile that does not show the code size statistics after compilation.  To show the ouput, there are two options:
- use the 'print-memory-usage' linker option
- use the memap.py tool for more detailed information

To get the output, I have modified the generated makefile:
1. adding print-memory-usage and generating the map file:
```
LD      = 'arm-none-eabi-gcc' '-Wl,--gc-sections' '-Wl,--wrap,main' '-Wl,--wrap,_malloc_r' '-Wl,--wrap,_free_r' '-Wl,--wrap,_realloc_r' '-Wl,--wrap,_calloc_r' '-Wl,--wrap,exit' '-Wl,--wrap,atexit' '-mcpu=cortex-m0plus' '-mthumb' '-Wl,--print-memory-usage' '-Wl,-Map=$(PROJECT).map'
```
2. calling the memap tool:
```
$(PROJECT).elf: $(OBJECTS) $(SYS_OBJECTS) $(LINKER_SCRIPT)
	+@echo "link: $(notdir $@)"
	@$(LD) $(LD_FLAGS) -T $(filter %.ld, $^) $(LIBRARY_PATHS) --output $@ $(filter %.o, $^) $(LIBRARIES) $(LD_SYS_LIBS)

	@echo ""
# for detailed memap, add option '-d'
	python .././mbed-os\tools\memap.py $(PROJECT).map -t GCC_ARM
	@echo ""

	@echo ""
	@echo "*****"
	@echo "***** You must modify vector checksum value in *.bin and *.hex files."
	@echo "*****"
	@echo ""
```

The 2nd method works only when a mbed-cli project including the tools is imported.

theotherjimmy, 2017-02-08:
So, one of the assumptions of the exporters is that the tools are not present on the system after the export. This is what happens on the online exporters (which use the same python). We could export the memap.py script into a "tools" directory to match the current setup, but that would require python on the system exported to.

theotherjimmy, 2017-02-08:
Option one is a good suggestion though. Do you know what that option would be for armlink and IAR?

JojoS62, 2017-02-08:
I don't have the IAR or armlink toolchains, so I don't know the options there. The exporter is for gcc_arm, is the same makefile used for the other tools?

theotherjimmy, 2017-02-08:
Yes.

JojoS62, 2017-02-08:
I got the export only working with a project generated with export to mbed-cli, there the tools are present. Is there now a way to get a project with mbed-os for a single MCU? And without all the features and tests stuff?

theotherjimmy, 2017-02-08:
That's what a website from the developer.mbed.org website looks like.

JojoS62, 2017-02-10:
ok, sorry. But the exporter story is still very annoying.
Two weeks ago, exporting from the online compiler a project with mbedLPC1768 for lpcxpresso worked, now it is broken again (an error occured during export, contact support).
Using eclipse_gcc_arm from the online compiler is new and can be used for lpcxpresso also, but with the big difference that it uses unmanaged build now.
Trying to export from a project page (https://developer.mbed.org/users/JojoS/code/OneButton/export) gives different options than the online compiler offers. Selectiing mbedLPC1768 and Eclipse-GCC-ARM gives another error:. 'Export failed: cannot concatenate 'str' and 'NoneType' objects'.
Using 'Import with CLI' creates a very large download, exporting to eclipse from here and importing to eclipse pulls in all avalaible targets and stuff and is also not practical now.

So, please what is the intended way now to get a working export to an eclipse based IDE?

JojoS62, 2017-02-10:
back to topic:
another option for the size info is callinig:
	arm-none-eabi-size $(PROJECT).elf
after linking.

theotherjimmy, 2017-02-10:
>. 'Export failed: cannot concatenate 'str' and 'NoneType' objects'.

So that's actually a bug with the website right now related to https://github.com/ARMmbed/mbed-os/issues/3728

theotherjimmy, 2017-02-10:
I like the calling size recommendation!

theotherjimmy, 2017-04-26:
@JojoS62 Would you like to make the Pull request to make this change? Would you rather I do it? (It may take some time, as it's a feature request)

closed 2018-06-06 by JojoS62