ARMmbed/mbed-os #3731: Exporting to IAR from online compiler results in missing references to top level files

maclobdell, 2017-02-09:
### Description
- Type: Bug 
- Related issue: n/a
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510, maybe more

**Toolchain:**
IAR, maybe more

**Toolchain version:**
n/a
**mbed-cli version:** 
n/a

**meed-os sha:**
mbed OS version 5.3.4

**DAPLink version:**
n/a

**Expected behavior**
export mbed-os-example-blinky , open in IDE, compile passes.

**Actual behavior**
export mbed-os-example-blinky to an IDE, open in IDE, compile fails because it is missing reference to main.cpp and mbed_config.h.  the files are there, but IAR doesn't know about them.

workaround: manually add those files to the project explorer and it works.

**Steps to reproduce**
export mbed-os-example-blinky from online compiler to an IAR IDE, open in IAR IDE, compile

0xc0170, 2017-02-09:
cc @theotherjimmy

theotherjimmy, 2017-02-10:
I think this is actually unrelated to the other problem you saw on Wednesday Mac.

theotherjimmy, 2017-04-26:
The tools release of 5.4.0 should have resolved this issue. Closing. please re-open if it's still a problem.

closed 2017-04-26 by theotherjimmy