ARMmbed/mbed-os #3735: STM stm_spi_api.c:spi_frequency has a bug with a -1 index into the baudrate_prescaler_table

helmut64, 2017-02-09:
The problem occures e.g. on the STM32L4 when a 40 Mhz hz value is being used. 
See file stm_spi_api.c:
>> handle->Init.BaudRatePrescaler = baudrate_prescaler_table[prescaler_rank-1];

With the 40000000 hz sample, the baudrate_prescaler_table is zero and initializes the Init.BaudRatePrescaler with random values. The previous hal implementation in spi_frequency file spi_api.c does not have this problem.
Regards Helmut

0xc0170, 2017-02-13:
cc @bcostm @adustm  @LMESTM @jeromecoutant

LMESTM, 2017-02-13:
I'll take it ...

helmut64, 2017-02-13:
Should be easy just to check if prescaler_rank is 0 than change it to 1.

The question I have, is the max SPI clock rate according to spi_get_clock_freq is 40 Mhz according to spi_get_clock_freq(), using the 0 scale divides it by 2 which means 20 Mhz.

Is this the max we can get from the SPI on the L4?
Or is the HAL_RCC_GetPCLK2Freq() not running fast enough?

I use a WinBond SPI Flash which can go as high as 104 Mhz, other models can go even up to 208 Mhz.

LMESTM, 2017-02-13:
@helmut64 

hello - first of all, thanks for reporting the issue. 

As of now, the PCLK2 freq is set to 40 MHz, which leaves 20MHz as the highest SPI frequency.

Though the PCLK2 could be increased from 40MHz to 80MHz (and spi from 20 to 40) by changing the below line with RCC_HCLK_DIV1 instead of RCC_HCLK_DIV2;
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/TARGET_NUCLEO_L476RG/system_stm32l4xx.c#L572

I will have to check with by colleague @jeromecoutant  to understand whether we can change the default setting. Until then, you can give it a try for your SPI flash usage.

LMESTM, 2017-02-13:
@helmut64 I tried to make the code more comprehensive as I pushed the fix. Please let me know if you can give it a try.

helmut64, 2017-02-14:
It works now fine for me, see my comments in #3759
Thank you for enhancing it.

LMESTM, 2017-02-15:
@helmut64  thank you for your prompt feedback ! Cheers

c1728p9, 2017-02-27:
@LMESTM I'm no longer able to set the SPI clock to 100KHz on multiple boards.

LMESTM, 2017-02-28:
@c1728p9 I'm proposing here #3759 to move error call to warn instead - let me know if ok

c1728p9, 2017-02-28:
@LMESTM sounds good to me.

closed 2017-03-09 by 0xc0170