ARMmbed/mbed-os #3744: Exporter: generated gcc_arm makefile calls linker with LD_FLAGS twice

JojoS62, 2017-02-10:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Bug

---------------------------------------------------------------
## Bug

**Target**
all

**Toolchain:**
GCC_ARM

**Actual behavior**
Exporter gcc_arm and eclipse_gcc_arm create a makefile. This makefile contains the options for the linker two times:
1st in the LD linker command, 2nd in the LD_FLAGS:
```
LD      = 'arm-none-eabi-gcc' '-Wl,--gc-sections' '-Wl,--wrap,main' '-Wl,--wrap,_malloc_r' '-Wl,--wrap,_free_r' '-Wl,--wrap,_realloc_r' '-Wl,--wrap,_calloc_r' '-Wl,--wrap,exit' '-Wl,--wrap,atexit' '--specs=nano.specs' '-mcpu=cortex-m0plus' '-mthumb' 

LD_FLAGS :=-Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit --specs=nano.specs -mcpu=cortex-m0plus -mthumb 
 ```

Then the linker is called:
@$(LD) $(LD_FLAGS)

For most flags the double definition is ignored (or better the last one wins), but for the --specs definition the linker complains about already specified specs.

**Steps to reproduce**
use export from online compiler to make_gcc or eclipse_gcc_arm project.

theotherjimmy, 2017-02-10:
Agreed. I'll provide a patch real quick. Also, I think this affects `make_*` and `eclpise_*`.

theotherjimmy, 2017-02-10:
There you go @JojoS62 #3745 should do the trick. Thanks for reporting this bug!

JojoS62, 2017-02-10:
thanks for the fast response!

please one more question: who and when should usualy close an issue?

theotherjimmy, 2017-02-10:
The person who merges the PR that resolves this issue should close it (automatically because of the Resolves line), or if the issue is fixed on master currently, then the person who opened it is supposed to close it.

PR #3745 (2017-02-17): Pass linker flags to the linker only once
Before, `make_gcc_arm` would export a makefile with both linker flags in
the `$(LD)` make variable and the `$(LD_FLAGS)` make variable. This only
affected GCC based makefiles, as the `$(LD)` make variable would only
contain the linker for both ARMC5 and IAR. This patch enforces the
`$(LD)` behavior of ARMC5 and IAR on all of the compilers by truncating
`$(LD)` to just the command.

Resolves #3744

## TODO
- [x] /morph export-build
closed 2017-02-17 by sg-