ARMmbed/mbed-os #3746: Mbed detect does not work on Windows

theotherjimmy, 2017-02-10:
```
>mbed detect
[mbed] Auto-installing missing Python modules...
Traceback (most recent call last):
  File "C:\Development\mbed-os\tools\detect_targets.py", line 77, in main
    if re.match(mcu_filter, mut['mcu']):
  File "c:\yotta\workspace\lib\re.py", line 141, in match
    return _compile(pattern, flags).match(string)
TypeError: expected string or buffer
[ERROR] expected string or buffer
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Development\mbed-os\tools\detect_targets.py" in "C:\Development\mbed-os"
```

Compare with linux:
```
$ mbed detect
[mbed] Detected K82F, port /dev/ttyACM0, mounted /mnt/FRDM-K82FD
[mbed] Supported toolchains for K82F
+--------+-----------+-----------+-----------+-----------+-----------+
| Target | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |
+--------+-----------+-----------+-----------+-----------+-----------+
| K82F   | Supported | Supported | Supported | Supported | Supported |
+--------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 3
```

pmancele, 2017-02-21:
Hello @theotherjimmy ,
We get mbed working both on linux and mac.
It may be your python version on windows which isnt compatible. We are using python 2.7. 
But to avoid modifying the python version of the system, we created virtualenv for mbed-os which use python 2.7. You should do that too if you plan to use different versions of python executable

theotherjimmy, 2017-04-26:
I can't reproduce this anymore.

closed 2017-04-26 by theotherjimmy